khoih-prog/EthernetWebServer #1: Support for STM32F Series

andy-hopkins, 2020-02-26:
Hi,

Your README says that From v1.0.2+, the library supports ... STM32F series .... But when I attempt to compile the PostServer example for an STM32F103, I get 

WARNING: library functional-vlpp claims to run on avr architecture(s) and may be incompatible with your current board which runs on STM32F1 architecture(s).

and subsequent complaints that PGM_VOID_P and memccpy_P aren't defined

Can you please confirm what STM32F boards are supported and/or if there are any simple defines I need to change to use the STM32F103

Cheers

Andy

Arduino IDE version: 1.8.12

khoih-prog, 2020-02-26:
The **EthernetWebServer** is written to be **somewhat** supportive to many different types of boards. But it also have to use other libraries, so the hardware support certainly depends on those external libraries.

Currently

1.  Just **ignore** the warning of **functional-vlpp** library. This is an external library I currently have no control. But I'll fix the warning by forking it then fix.

2. Ethernet library for **W5x00 doesn't support STM32F** boards. You can post an issue there to request _Feature Enhancement_. Whenever that library is supporting **STM32F** boards, this library will automatically support. If I have time, I'll fork and fix Ethernet library to provide support to **STM32F** series.

3. Currently, **EthernetWebServer** library can only support some **STM32F** boards running **ENC28J60** shields (only because it's using UIPEthernet library supporting STM32F). The prerequisite is the board must have **Flash larger than 128K**.

For example:
- STM32F103CB
- Microduino Core STM32
- Maple Mini
- Maple Rev3
- STM Nucleo
- STM32F103RB, RC and RE
- STM32F103TB
- STM32F103VC, VD and VE
- STM32F103Z
- HYTiny STM32F103TB

But if there is any issue with the **UIPEthernet** library, please post the issue there.

4. These boards are not supported even using **ENC28J60**
- STM32F103C8, R8, T8 having only 64K Flash
- STM32F4 as there is no HardwareSerial

In order to use supported STF32F1 boards, you have to declare this in eny sketch

```cpp
#define ETHERNET_USE_SAMD   true

// Use true  for ENC28J60 and UIPEthernet library (https://github.com/UIPEthernet/UIPEthernet)
// Use false for W5x00 and Ethernetx library      (https://www.arduino.cc/en/Reference/Ethernet)
#define USE_UIP_ETHERNET   true
```

khoih-prog, 2020-02-27:
**Update**

These boards running ENC28J60 (with Flash 64+K) are supported if using official STM32 core from STMicroElectronics:
[**Arduino_Core_STM32**](https://github.com/stm32duino/Arduino_Core_STM32)

 1. Nucleo-144
 2. Nucleo-64
 3. Discovery
 4. STM32MP1
 5. Generic STM32F1 (with 64+K Flash): C8 and up
 6. Generic STM32F4
 7. STM32L0
 8. LoRa boards
 9. 3-D printer boards
10. Generic Flight Controllers
11. Midatronics boards

and these are not supported:

1. Nucleo-32 (small Flash/memory)
2. Eval (no Serial, just need to redefine in sketch, library and **UIPEthernet**)
3. Generic STM32F0 (small Flash/memory)
4. Generic STM32F1 (with <64K Flash): C6
5. Generic STM32F3 : no _HardwareSPI.h_
6. Electronics Speed Controllers  (small Flash/memory)

closed 2020-02-27 by khoih-prog

khoih-prog, 2020-02-27:
Dear @andy-hopkins 

I just finished and published a new EthernetWebServer_STM32 library for STM32 boards. Please give it a try.

https://github.com/khoih-prog/EthernetWebServer_STM32

This library currently supports
1. STM32 boards with built-in Ethernet such as :
  - ***Nucleo-144 (F429ZI, F767ZI)***
  - ***Discovery (STM32F746G-DISCOVERY)***
2. ***STM32 boards (with 64+K Flash) running EMC28J60 shields***

This is simple yet complete WebServer library for `STM32` boards running built-in Ethernet (Nucleo-144, Discovery) or EMC28J60 Ethernet shields. The functions are similar and compatible to ESP8266/ESP32 WebServer libraries to make life much easier to port sketches from ESP8266/ESP32.

The library supports 
1. HTTP Server and Client
2. HTTP GET and POST requests, provides argument parsing, handles one client at a time.

Regards,
KH

andy-hopkins, 2020-02-27:
Hi @khoih-prog 

I have success with the PostServer example with a BluePill and ENC28J60.

No pressure, but any idea how long till, or if, the F407VET6 will be supported?

Cheers

khoih-prog, 2020-02-28:
Hi @andy-hopkins 

Good news that you're successful with BluePill and ECN28J60.

Please try the following new library and let us know the result for F407VET6 and all remaining STM32. BTW, I'm planning to port W5x00, etc. Ethernet shield to this new lib.

[**EthernetWebServer_STM32**](https://github.com/khoih-prog/EthernetWebServer_STM32)

All the STM32 boards with more than 64K Flash are supported by that lib, including all **F407VET**.
Don't use this library anymore if you're using STM32 as it's not as efficient as the new one (larger code size, **not standard core from STMicro**).

Boards supported by [**EthernetWebServer_STM32**](https://github.com/khoih-prog/EthernetWebServer_STM32) library currently supports

1. STM32 boards with **built-in Ethernet** such as :
  - Nucleo-144 (F429ZI, F767ZI)
  - Discovery (STM32F746G-DISCOVERY)
2. STM32 boards (with 64+K Flash) running **EMC28J60 shields**
  - Nucleo-144
  - Nucleo-64
  - Discovery
  - STM32MP1
  - Generic STM32F1 (with 64+K Flash): C8 and up
  - Generic STM32F4
  - STM32L0
  - LoRa boards
  - 3-D printer boards
  - Generic Flight Controllers
  - Midatronics boards

3. and these boards are not supported:

  -  Nucleo-32 (small Flash/memory)
  -  Eval (no Serial, just need to redefine in sketch, library and UIPEthernet)
  -  Generic STM32F0 (small Flash/memory)
  -  Generic STM32F1 (with <64K Flash): C6
  -  Generic STM32F3 : no HardwareSPI.h
  -  Electronics Speed Controllers (small Flash/memory)

Cheers,
KH