khoih-prog/EthernetWebServer #3: Problem with your custom ethernet library (W5500-Teensy 4.0)

adriansmas, 2020-04-24:
Hi! I'm working with your library with a teensy 4.0 and Ethernet Chip W5500. With default Ethernet library, I can't send messages longer than 2K. When I paste your custom files of that library I can't send any messages. "HTTP_MAX_CLOSE_WAIT" Timeout.
Do you have any suggestion of what may be my problem?

adriansmas, 2020-04-24:
Also gif image of "HelloServer2" cannot be sent with the default functions.

khoih-prog, 2020-04-24:
Did you test with the unmodified Ethernet library? Is it working OK with any simple example, such as HelloServer? 

Just be sure no hardware-related issue.

If OK, test the modified Ethernet library with the simplest example, such as HelloServer.
Can you also post your full code along with any terminal / debug messages?

closed 2020-04-30 by khoih-prog