khoih-prog/EthernetWebServer #5: Wemos D1 mini and W5500 not work in standart SS/CS pin

BorgMcz, 2020-05-05:
if I connect W5500 and Wemos D1 mini (esp8266) using the default settings (for CS pin a connection with a transistor according to: https://esp8266hints.wordpress.com/2018/02/13/adding-an-ethernet-port-to is used -your-esp-revisited /) the following libraries do not work:

- Ethernet (arduino v.2.0.0) - not work, + unexpected system restarts

- Ethernet2 (v.1.0.4) - not work

- Ethernet3 (v.1.5) - not work default cs/ss pin , after changing the pin to GPIO5 (wemos D1) it works fully (another suitable pin can be used). It is even possible to use the default pin, but it must be declared `Ethernet.setCsPin (15);`

khoih-prog, 2020-05-05:
Dear @BorgMcz 

The EthernetWebServer **depends on many other libraries**, such as Ethernet, Ethernet2, etc. and can not do the work alone if there is bug or connection is not correct or a new variation of board is used.

The best way to solve a similar issue is to run a simple example from the underlying libraries (Ethernet, Ethernet2, etc. )  to verify if the pin usage is working before using the higher level of libraries, such as EthernetWebServer.

If you still have the problem with the underlying libraries, it's better to open an issue there so that the issue could be solved completely at the correct place.

I still can investigate later, if the problem can be solved, can just provide you a quick-fixing kludge / patch for the library.

But as you known, **W5x00 CS/SS normally is pin 10**, and some Mega  / SAM DUE boards have the internal SPI bus pins assigned to pins 50, 51, 52, 53 or the ICSP connector so that W5100 can be used more easily.

Thanks for your sharing of experience.

Regards,

khoih-prog, 2020-05-06:
Dear @BorgMcz

I've just had some time to test and can duplicate your findings and solution that Ethernet3 lib can work OK with W5500 and ESP8266.

I used NodeMCU 1.0 and Ethernet.setCsPin (D1);

```
#if (defined(ETHERNET_USE_ESP8266) && USE_CUSTOM_ETHERNET)
  // For Ethernet3, OK with D0(GPIO16), D1(GPIO5), D2(GPIO4), D3(GPIO0)
  // , D4(2), D8() not OK
  Ethernet.setCsPin (D1);
#endif
```
and the terminal output

```
Starting AdvancedServer on ESP8266
=========================
Used/default SPI pinout:
MOSI:13
MISO:12
SCK:14
SS:15
SPI_CS:16  <==== D0
=========================
HTTP EthernetWebServer is @ IP : 192.168.2.79
...
Starting AdvancedServer on ESP8266
Used/default SPI pinout:
MOSI:13
MISO:12
SCK:14
SS:15
SPI_CS:5            <==== D1
=========================
HTTP EthernetWebServer is @ IP : 192.168.2.79
...
Starting AdvancedServer on ESP8266
=========================
Used/default SPI pinout:
MOSI:13
MISO:12
SCK:14
SS:15
SPI_CS:4   <=== D2
=========================
HTTP EthernetWebServer is @ IP : 192.168.2.79
...
Starting AdvancedServer on ESP8266
=========================
Used/default SPI pinout:
MOSI:13
MISO:12
SCK:14
SS:15
SPI_CS:0   <=== D3
=========================
HTTP EthernetWebServer is @ IP : 192.168.2.79
...
Starting AdvancedServer on ESP8266
=========================
Used/default SPI pinout:
MOSI:13
MISO:12
SCK:14
SS:15
SPI_CS:2    <=== D4 not OK
=========================
HTTP EthernetWebServer is @ IP : (IP unset)
```
I think it's necessary to reopen this issue and together find a final solution for ESP8266 and W5x00. We might possibly find a solution then post a library patch so that everybody can benefit from the patch.
I believe with the findings with Ethernet3 lib, we can know how to fix the other libs, such as Ethernet, EthernetLarge, Ethernet2, Arduino Ethernet, ESP8266 Ethernet
Please post more test results and fixes so that we can publish the patch.

Many thanks for your findings.

Regards,

khoih-prog, 2020-05-13:
Just fix the issue in v1.0,8. Now ESP8266/ESP32 can run OK with Ethernet, Ethernet2, Ethernet3 and EthernetLarge libraries.

closed 2020-05-13 by khoih-prog