khoih-prog/EthernetWebServer #6: EthernetWebServer does not read and send html.h file to web browser

EY789, 2020-05-16:
Hello Khoih,

I am trying to read a file test.h containing html code to send it to the web browser but shows a blank page.

`#include test.h

void handleRootl()
{
  String root_code = html_page;
  server.send(200,"text/html",root_code); 
}`

test.h file contains:

`const char html_page[] PROGMEM = R"=====(
<!DOCTYPE HTML>

<html> ........ </html>
)=====";`

The sketch works fine with ESP8266WebServer. Please let me know how to solve it.

khoih-prog, 2020-05-16:
Because you're using resource-limited Mega, not powerful ESP8266, with powerful libraries to support complicated features.

Moreover, the W5x00 shields also have 2K buffer size limitation unless you change.

Try the smallest html page, then add more until it breaks. But the html page size must be much less than 2K. Must be very careful while writing code to avoid unnecessary large local vars, such as 

```void handleRoot()
{
  String root_code = html_page;  <== create big unnecessary local var.
  server.send(200,"text/html",root_code);
}
```

Start with the following code

```
void handleRoot()
{
  server.send(200,"text/html", html_page);
}
```

test.h
```

const char html_page[] = "<!DOCTYPE html><html><head><title>AVR_WM</title><style>div,input{padding:5px;font-size:1em;}input{width:95%;}body{text-align: center;}\
fieldset{border-radius:0.3rem;margin:0px;}</style></head>\
<div style=\"text-align:left;display:inline-block;min-width:260px;\"><fieldset>\
<div><label>SSID</label><input value=\"your-SSID\"id=\"id\"><div></div></div>\
<div><label>PWD</label><input value=\"your-PWD\"id=\"pw\"><div></div></div></fieldset>";
```

and have this webpage displayed

[image]

closed 2020-05-16 by khoih-prog

EY789, 2020-05-17:
Thanks for clarifying. Indeed the test.h file size is 17KB so saved it on SD card and loaded it in portions to the web browser within the available resources limitations.

innsbruckcreate, 2020-11-11:
Hi Khoih,
I am facing the same problem. I am using Arduino due and ethernet shield 2, my Html page consists of JavaScript and CSS files and is big. The EthernetWebServer does not read and send html file  to web browser but if i remove the JavaScript and CSS, the Ethernet Webserver is able to read the html file. I do not want to use SD card in my project although it is working with sd card. Please help me in this regard.

khoih-prog, 2020-11-11:
@innsbruckcreate 

The Ethernet shields, using Ethernetx libraries with 2K-buffer limitations, as well as the small MCUs, all have limited resources. Don't expect them working like a modern PC.

Please have a look at [**Issue 7: Problem with page larger than 2k.**](https://github.com/khoih-prog/EthernetWebServer/issues/7) to see how the well-known big-file issue was solved.

> Thanks for the advice, I finally solved the problem by separating the css style into the file itself and then a separate html file.

innsbruckcreate, 2020-11-12:
Hi Khoih
Thanks for the response.
I have tried with simple file (for example analog gauge reading or bar graph) with 1kb size, but then also Ethernet web server is is not able to read the file it gives a blank webpage.
[index.txt](https://github.com/khoih-prog/EthernetWebServer/files/5530567/index.txt)
Thanks
Regards
Durgesh Singh

khoih-prog, 2020-11-12:
Hi,

I'm afraid I can't help any further as I don't have time to deal with your specific problem.

This [**EthernetWebServer Library**](https://github.com/khoih-prog/EthernetWebServer) has to rely on `Ethernetxyz libraries` and can't provide magic.

I suggest you first try to successfully use the `Ethernetxyz libraries`, and **ask for help and support there and/or in any of other related support forums**. Remember also to provide enough info as described in [**How to create a Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) so that other willing-to-help people can duplicate te issue and help.

Whenever you're successful in using the `Ethernetxyz libraries` with your use case, then move to use this [**EthernetWebServer Library**](https://github.com/khoih-prog/EthernetWebServer).

Good Luck,