khoih-prog/EthernetWebServer #7: Problem with page larger than 2k.

BorgMcz, 2020-05-24:
Hello, I don't know exactly where the problem is, but if I try to create an html page larger than 2k, a part of the page will load in the browser (corresponds to a size of about 2k) and then restart for an internal error. I use W5500 (ethernet3), esp8266 (sdk 2.7.1), ethernetwebserver (v 1.0.7).

`void handleRoot()
{
  char temp[3000];
  uptime::calculateUptime();
  float x1 = newVoltageSend;
  int x2 = uptime::getDays();
  int x3 = uptime::getHours();
  int x4 = uptime::getMinutes();
  int x5 = uptime::getSeconds();
  
snprintf(temp, 3000, 
"<html><head><meta http-equiv='refresh' content='5'/>\
<title>ESP8266 WebMonitor - Status info</title>\
<style>body{text-align:center;}table.redTable{font-family:Arial,Helvetica,sans-serif;border:2px solid #A40808;background-color:#EEE7DB;width:330px;height:250px;text-align:center;border-collapse:collapse;margin-left:auto;margin-right:auto;}\
table.redTable td,table.redTable th{border:1px solid #AAAAAA;padding:3px 2px;}\
table.redTable tbody td{font-size:15px;font-weight:bold;}\
table.redTable tr:nth-child(even){background:#F5E1E0;}\
.......
<tr><td><a href='/resend/'>Resend</a></tr>\
<tr><td><a href='/reboot/'>Reboot</a></td></tr>\
<tr><td><a href='/start/'>Upgrade</a></td></tr>\</table>\
</body></html>",
  x1, x2, x3, x4, x5);
  server.send(200, "text/html", temp);
}
`

I initialize ethernet according to the procedure from the example to the size 4k, I also tried a bigger one.
` // Use  MAX_SOCK_NUM = 4 for 4K, 2 for 8K, 1 for 16K RX/TX buffer
  #define ETHERNET3_MAX_SOCK_NUM     4
  Ethernet.init (ETHERNET3_MAX_SOCK_NUM);
    Ethernet.begin(eeprom_config.mac, ip, subnet, gateway, dns_server);
`

I don't know where I can make a mistake, can you help me how to send a larger page as 2k.
Thank you

khoih-prog, 2020-05-25:
Can you try with the latest [EthernetWebServer v1.0.9](https://github.com/khoih-prog/EthernetWebServer) and use Ethernet, Ethernet2, Ethernet3 , or EthernetLarge libraries to see if it's OK. Also change the CS/SS pin to D2.

I'm afraid you've got into the notorious 2K buffer issue of Ethernetx libraries and it can be solved only if you open an issue at the related Ethernetx library. The [EthernetWebServer library](https://github.com/khoih-prog/EthernetWebServer) is just relying and using those Ethernetx libs and can not fix the real issue.

I suggest if your problem persists, you can use the ESP8266 WiFi to do those large html tasks.

closed 2020-05-25 by khoih-prog

BorgMcz, 2020-05-25:
Good day. I get this answer from the creator of the ethernet3 library.

> The max size for data in an Ethernetframe is generell 1500 bytes. The w5500 have no jumbo pakets. The socket RAM is only a buffer for messages. You avoid the lost of pakets when there is heavy traffic. You must divide the message in parts.<

Is it possible to give an example of how I can split a message (html page)? It would help me a lot.

THX

khoih-prog, 2020-05-26:
Try to use either these libs with the latest [**EthernetWebServer v1.0.9**](https://github.com/khoih-prog/EthernetWebServer) with **2K+ buffer**

1. [Ethernet library](https://www.arduino.cc/en/Reference/Ethernet)
2. [EthernetLarge library](https://github.com/OPEnSLab-OSU/EthernetLarge)

Also remember to use the libraries' fixes mentions in to make 2K+ buffer ready to use in [Important Notes](https://github.com/khoih-prog/EthernetWebServer#important-notes)

You can also have a look at how the **2K+ html pages can be made and loaded** in file [BlynkEthernet_WM.h](https://github.com/khoih-prog/BlynkEthernet_WM/blob/master/src/Adapters/BlynkEthernet_WM.h) of [BlynkEthernet_WM Library](https://github.com/khoih-prog/BlynkEthernet_WM)

Try to **avoid large local variables** (such as html string) as stack size is limited.

It's also advisable to use **ESP32/ESP8266** for that purpose to avoid headaches with large html files.

BorgMcz, 2020-05-26:
Thanks for the advice, I finally solved the problem by separating the css style into the file itself and then a separate html file.

This way, hopefully there will be enough space for the necessary page.

Unfortunately, using modified libraries caused me a problem following the ICMP ping library.

> Alway pass large variables, created as global vars, as reference
You can give me an example, I'm not sure how to do it right here.

Thanks again.

khoih-prog, 2020-05-27:
Always pass large variables, created as global vars, as reference

For example

1. Worst, will cause issue

```
String createHTML(void)
{
  String localLargeHTMLString = "very large string of more than 2K";
  
  localLargeHTMLString += "add lot more stuffs here";
  
  return localLargeHTMLString;
}

void sendHTML(void)
{
  String html = createHTML();
  
  // Display here to check if html is OK before sending
  Serial.println(html);
  
  server->send(200, "text/html", html);
  ....
}

```

or 

```
String globalLargeHTMLString = "very large string of more than 2K";

String createHTML(void)
{
  String localLargeHTMLString;
   
  localLargeHTMLString = globalLargeHTMLString + "add lot more stuffs here";
  
  return localLargeHTMLString;
}

void sendHTML(void)
{
  String html = createHTML();
  
  // Display here to check if html is OK before sending
  Serial.println(html);
  
  server->send(200, "text/html", html);
  ....
}

```

2. Also can cause issue

```
String globalLargeHTMLString = "very large string of more than 2K";

String createHTML(void)
{ 
  globalLargeHTMLString += "add lot more stuffs here";
  
  return globalLargeHTMLString;
}

void sendHTML(void)
{
  String html = createHTML();
  
  // Display here to check if html is OK before sending
  Serial.println(html);
  
  server->send(200, "text/html", html);
  ....
}

```

3. Better way

```
String globalLargeHTMLString = "very large string of more than 2K";

void createHTML(String &inputString)
{
  inputString += "add lot more stuffs here";
  
  return;
}

void sendHTML(void)
{
  createHTML(globalLargeHTMLString);
  
  // Display here to check if html is OK before sending
  Serial.println(globalLargeHTMLString);
  
  server->send(200, "text/html", globalLargeHTMLString);
  ....
}

```