khoih-prog/EthernetWebServer #8: Arduino hangs after repeated refreshing

prenone, 2020-07-17:
With the following code

```
#include <Ethernet.h>
#include <EthernetWebServer.h>

#define webpage "Big HTML string" //2114 bytes. I have applied the patch.

EthernetWebServer webServer;
void setup()
{
  SerialUSB.begin(9600);

  Ethernet.begin(settings.macAddress, settings.ipAddress, settings.ipAddressDns, settings.ipAddressGateway, settings.ipAddressSubnet); // 'settings' is a struct declared elsewhere.
  webServer = EthernetWebServer(settings.portHttp);

  webServer.on("/login", [&webServer]{ // /login handler
    webServer.send(200, "text/html", webpage);
  });
}

void loop()
{
  webServer.handleClient();
}
```
When I refresh the page multiple times the Arduino freezes completely (not just the Web Server, I have verified by placing SerialUSB.println in the loop()). I assume this is because the big string is loaded into memory but never unloaded. I have also tried encapsulating the big string in the F() but I get the same result. Any idea of why this is or how to solve this?

khoih-prog, 2020-07-19:
Hi @prenone 

Thanks for using the library.

I test the following code with **SAM DUE** (SAM3X8E) and **W5100** Ethernet with the html size of **2211** bytes.
Refreshing the webbrowser hundred times without any issue.
I also turned on the **_ETHERNET_WEBSERVER_LOGLEVEL_** to 4 to have mode debug messages.

I'm afraid you have some problem with your **code, W5500 hardware or Zero is too slow** for your application. Only you can find out where the issue is. Good luck.

1. SAM DUE, 84MHz, 96KB SRAM, 512 KB Flash
2. ZERO, 48MHz, 32  KB SRAM, 256 KB Flash

[image]

```
#define _ETHERNET_WEBSERVER_LOGLEVEL_     4

#include <Ethernet.h>
#include <EthernetWebServer.h>

//#define webpage "Big HTML string"

const char webpage[] = "<!DOCTYPE html><html><head><title>Blynk_Esp32_BT_BLE_WF</title><style>div,input{padding:2px;font-size:1em;}input{width:95%;}\
body{text-align: center;}button{background-color:#16A1E7;color:#fff;line-height:2.4rem;font-size:1.2rem;width:100%;}fieldset{border-radius:0.5rem;margin:0px;}\
</style></head><div style=\"text-align:left;display:inline-block;min-width:260px;\">\
<fieldset><div><label>WiFi SSID</label><input value=\"[[id]]\"id=\"id\"><div></div></div>\
<div><label>PWD</label><input value=\"[[pw]]\"id=\"pw\"><div></div></div>\
<div><label>WiFi SSID1</label><input value=\"[[id1]]\"id=\"id1\"><div></div></div>\
<div><label>PWD1</label><input value=\"[[pw1]]\"id=\"pw1\"><div></div></div></fieldset>\
<fieldset><div><label>Blynk Server</label><input value=\"[[sv]]\"id=\"sv\"><div></div></div>\
<div><label>WiFi Token</label><input value=\"[[tk]]\"id=\"tk\"><div></div></div>\
<div><label>Blynk Server1</label><input value=\"[[sv1]]\"id=\"sv1\"><div></div></div>\
<div><label>WiFi Token1</label><input value=\"[[tk1]]\"id=\"tk1\"><div></div></div>\
<div><label>Port</label><input value=\"[[pt]]\"id=\"pt\"><div></div></div></fieldset>\
<fieldset><div><label>WiFi SSID</label><input value=\"[[id]]\"id=\"id\"><div></div></div>\
<div><label>PWD</label><input value=\"[[pw]]\"id=\"pw\"><div></div></div>\
<div><label>WiFi SSID1</label><input value=\"[[id1]]\"id=\"id1\"><div></div></div>\
<div><label>PWD1</label><input value=\"[[pw1]]\"id=\"pw1\"><div></div></div></fieldset>\
<fieldset><div><label>Blynk Server</label><input value=\"[[sv]]\"id=\"sv\"><div></div></div>\
<div><label>WiFi Token</label><input value=\"[[tk]]\"id=\"tk\"><div></div></div>\
<div><label>Blynk Server1</label><input value=\"[[sv1]]\"id=\"sv1\"><div></div></div>\
<div><label>WiFi Token1</label><input value=\"[[tk1]]\"id=\"tk1\"><div></div></div>\
<div><label>Port</label><input value=\"[[pt]]\"id=\"pt\"><div></div></div></fieldset>\
<fieldset><div><label>BT Token</label><input value=\"[[bttk]]\"id=\"bttk\"><div></div></div>\
<div><label>BLE Token</label><input value=\"[[bltk]]\"id=\"bltk\"><div></div></div></fieldset>\
<fieldset><div><label>Board Name</label><input value=\"[[nm]]\"id=\"nm\"><div></div></div></fieldset>";

IPAddress ipAddress(192, 168, 2, 200);

IPAddress ipAddressDns(192, 168, 2, 1);
IPAddress ipAddressGateway(192, 168, 2, 1);
IPAddress ipAddressSubnet(255, 255, 255, 0);

uint16_t portHttp = 80;

byte macAddress[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x09 };

EthernetWebServer webServer(portHttp);

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.println("\nStarting WebServer");

  //Ethernet.begin(settings.macAddress, settings.ipAddress, settings.ipAddressDns, settings.ipAddressGateway, settings.ipAddressSubnet); // 'settings' is a struct declared elsewhere
  Ethernet.begin(macAddress, ipAddress, ipAddressDns, ipAddressGateway, ipAddressSubnet); 
  //webServer = EthernetWebServer(settings.portHttp);
[... 706 lines cut ...]
headerValue:  192.168.2.200
headerName:  Connection
headerValue:  keep-alive
headerName:  Cache-Control
headerValue:  max-age=0
headerName:  DNT
headerValue:  1
headerName:  Upgrade-Insecure-Requests
headerValue:  1
headerName:  User-Agent
headerValue:  Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.89 Safari/537.36
headerName:  Accept
headerValue:  text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
headerName:  Accept-Encoding
headerValue:  gzip, deflate
headerName:  Accept-Language
headerValue:  en-GB,en-US;q=0.9,en;q=0.8
args:  
Request:  /login
Arguments:  
send1: len =  2079
content =  <!DOCTYPE html><html><head><title>Blynk_Esp32_BT_BLE_WF</title><style>div,input{padding:2px;font-size:1em;}input{width:95%;}body{text-align: center;}button{background-color:#16A1E7;color:#fff;line-height:2.4rem;font-size:1.2rem;width:100%;}fieldset{border-radius:0.5rem;margin:0px;}</style></head><div style="text-align:left;display:inline-block;min-width:260px;"><fieldset><div><label>WiFi SSID</label><input value="[[id]]"id="id"><div></div></div><div><label>PWD</label><input value="[[pw]]"id="pw"><div></div></div><div><label>WiFi SSID1</label><input value="[[id1]]"id="id1"><div></div></div><div><label>PWD1</label><input value="[[pw1]]"id="pw1"><div></div></div></fieldset><fieldset><div><label>Blynk Server</label><input value="[[sv]]"id="sv"><div></div></div><div><label>WiFi Token</label><input value="[[tk]]"id="tk"><div></div></div><div><label>Blynk Server1</label><input value="[[sv1]]"id="sv1"><div></div></div><div><label>WiFi Token1</label><input value="[[tk1]]"id="tk1"><div></div></div><div><label>Port</label><input value="[[pt]]"id="pt"><div></div></div></fieldset><fieldse [... cut ...] iv></div><div><label>WiFi Token1</label><input value="[[tk1]]"id="tk1"><div></div></div><div><label>Port</label><input value="[[pt]]"id="pt"><div></div></div></fieldset><fieldset><div><label>BT Token</label><input value="[[bttk]]"id="bttk"><div></div></div><div><label>BLE Token</label><input value="[[bltk]]"id="bltk"><div></div></div></fieldset><fieldset><div><label>Board Name</label><input value="[[nm]]"id="nm"><div></div></div></fieldset>
[ETHERNET_WEBSERVER] handleClient:Client connected

```

closed 2020-07-19 by khoih-prog

prenone, 2020-07-19:
Hello, thank you for your prompt response. 
I managed to solve the problem I was having by initializing the Serial with Serial.begin(). I have an ESP8266 connected to that Serial. I don't know why not having Serial initialized (I am not even using the ESP8266 and I have no libraries for it installed) makes the processor crash after a couple of requests though.