khoih-prog/EthernetWebServer #13: Dependencies in library.json

Phintly, 2020-11-25:
Why is the functional library not listed in library.json under dependencies so that PlatformIO can install it automatically?

khoih-prog, 2020-11-25:
It's been done in both

1. [**platformio.ini**](https://github.com/khoih-prog/EthernetWebServer/blob/master/platformio/platformio.ini) 

```
lib_deps = 
; PlatformIO 4.x 
  Functional-Vlpp@~1.0.1
  Ethernet@~2.0.0
  EthernetLarge@~2.0.0
  Ethernet2@~1.0.4
  Ethernet3@~1.5.3
  EthernetENC@~2.0.0
  UIPEthernet@~2.0.9
; PlatformIO 5.x  
;  khoih-prog/Functional-Vlpp@~1.0.1
;  PaulStoffregen/Ethernet@~2.0.0
;  PaulStoffregen/EthernetLarge@~2.0.0
;  adafruit/Ethernet2@~1.0.4
;  sstaub/Ethernet3@~1.5.3
;  jandrassy/EthernetENC@~2.0.0  
;  UIPEthernet/UIPEthernet@~2.0.9
```
and

2. [**library.properties**](https://github.com/khoih-prog/EthernetWebServer/blob/master/library.properties)

```
depends=Functional-Vlpp
```

It's working with other libraries. If it doesn't work for you, either check your PIO installation or post a question on [PIO Forum](https://community.platformio.org/)

closed 2020-11-25 by khoih-prog

Phintly, 2020-11-25:
Manually copying the library's dependencies into your own dependencies contradicts the purpose of a dependency management system and library.properties is the configuration for the standard library manager, not for PlatformIO. See: https://docs.platformio.org/en/latest/librarymanager/config.html#dependencies

khoih-prog, 2020-11-25:
PIO is new and is **changing quickly**. The library original design is for Arduino IDE which relies only on `library.properties`.
Many libraries' dependency installations, so far, are working OK with PIO, but some not. Ask PIO why.

But anyway I'll **add the dependencies, from now on, into all new libraries** to support this PIO feature. Hopefully will be OK for another while.

Thanks,

Phintly, 2020-11-25:
Are you going to add the dependency to this library in the next version too? The wording is a bit unclear.

Phintly, 2020-11-25:
Nvm I haven't seen the commit

khoih-prog, 2020-11-25:
Already updated on master of this lib. 
Certainly will add to every libraries I'm currently maintaining.

khoih-prog, 2020-11-25:
The way PIO is doing is very clumsy.

It'll take a lot of time to search and know the correct "owner" and "name" of each dependent library. Many times, the library's name is different from the one used in Arduino IDE.

---

For example:

[**STM32duino STM32Ethernet Library**](https://github.com/stm32duino/STM32Ethernet)

1. Arduino Library name: 

```
name=STM32duino STM32Ethernet
...
depends=STM32duino LwIP
```

2. PIO changes that to

```
{
   "owner": "stm32duino",
   "name": "STM32Ethernet",
   "version": "^1.2.0"
}
```

and you can only know that when you search, then access access the Library in PIO and see

```
pio lib install "stm32duino/STM32Ethernet"
```
---

That's why many libraries writers don't care about `library.json` and PIO. Just only `library.properties`, even nothing (no `library.json`  and `library.properties`).

You have to try to add libraries' dependencies once and understand the issue.

khoih-prog, 2020-11-25:
Then for 

[**STM32duino LwIP Library**](https://github.com/stm32duino/LwIP)

1. Arduino Library name: 

```
name=STM32duino LwIP
```

2. PIO changes that to

```
{
   "owner": "stm32duino",
   "name": "STM32duino LwIP",      <=========== now using original IDE name
}
```
and you can only know that when you search, then access access the Library in PIO [**STM32duino LwIP**](https://platformio.org/lib/show/1788/STM32duino%20LwIP)

and see

```
pio lib install "stm32duino/STM32duino LwIP"
```

Phintly, 2020-11-26:
Oh god, you are still working on it. I didn't mean to waste your time like that. Maybe try just removing the `library.json` altogether in the libraries you haven't yet supplemented by the dependencies since, like I said, PIO should also work with the `library.properties` afaik.

> The only requirement is a library manifest file - library.json, library.properties, or module.json.
https://docs.platformio.org/en/latest/librarymanager/creating.html

Or you could just wait until someone complains.

khoih-prog, 2020-11-26:
Thanks for your words.

Just try my best to save you all a lot of time later.

In the lengthy and time-consuming process to modify all 50+ libs. Hopefully once and for all. ;-((