khoih-prog/EthernetWebServer #14: Compile Error on Teensy 4.1

jimmie11, 2020-12-03:
Thank you for your efforts.  I was very happy to find a library that works on Teensy 4.1 however I am unable to compile the WebServer example.  I had download the VLP library and also copied boards.txt but this did not help.  I also tried changing Ethernet.h to NativeEthernet.h but his did not help either.

Here is the error:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/Stream.h:68:6: error: 'int Stream::timedRead()' is private

  int timedRead();

      ^

C:\Users\TTC\Documents\Arduino\libraries\EthernetWebServer-master\src\Ethernet_HTTPClient\Ethernet_HttpClient.cpp:642:13: error: within this context

     int c = timedRead();

             ^

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/HardwareSerial.h:106:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/WProgram.h:46,

                 from C:\Users\TTC\AppData\Local\Temp\arduino_build_394779/pch/Arduino.h:6,

                 from C:\Users\TTC\Documents\Arduino\libraries\EthernetWebServer-master\src/Ethernet_HTTPClient/Ethernet_HttpClient.h:43,

                 from C:\Users\TTC\Documents\Arduino\libraries\EthernetWebServer-master\src\Ethernet_HTTPClient\Ethernet_HttpClient.cpp:44:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/Stream.h:68:6: error: 'int Stream::timedRead()' is private

  int timedRead();

      ^

C:\Users\TTC\Documents\Arduino\libraries\EthernetWebServer-master\src\Ethernet_HTTPClient\Ethernet_HttpClient.cpp:642:23: error: within this context

     int c = timedRead();

                       ^

Multiple libraries were found for "Ethernet.h"
 Used: C:\Program Files (x86)\Arduino\libraries\Ethernet
Not used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Ethernet-master
Error compiling for board Teensy 4.1.

jimmie11, 2020-12-03:
I made the following changes and it is now working.

#define USE_ETHERNET          false 
//....
//....
#define USE_CUSTOM_ETHERNET   true

#elif USE_CUSTOM_ETHERNET
   #include "NativeEthernet.h"

closed 2020-12-03 by jimmie11

khoih-prog, 2020-12-03:
Thanks for your interest in the library and the solution. I'll include it in the next release.
It'll definitely be very helpful for other Teensy users.

jimmie11, 2020-12-03:
Hello Khoi,

Thank you for your efforts in making your library available.

I forgot to mention one more change that is needed:

In:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4\Stream.h

Change from private to public.

  public:
        char read_error;
        int timedRead();
        int timedPeek();
        int peekNextDigit();

Regards.

khoih-prog, 2020-12-03:
It'll be great if you can post your experience on Teensy Forum to help Teensy users there.

[**Teensy 4.1 NativeEthernet Library**](https://forum.pjrc.com/threads/60857-T4-1-Ethernet-Library)

BR

jimmie11, 2020-12-04:
I posted the info at this link:

https://forum.pjrc.com/threads/64953-Teensy-4-1-Ethernet-Problems?p=262054#post262054

khoih-prog, 2020-12-04:
Hi @jimmie11 

Just release [**EthernetWebServer v1.2.0**](https://github.com/khoih-prog/EthernetWebServer/releases/tag/v1.2.0)

Can you test to see if there is any more issue and give feedback.

Your contribution is noted in [**Contributions-and-Thanks**](https://github.com/khoih-prog/EthernetWebServer#contributions-and-thanks)

---

### Release v1.2.0

1. Add support to [**NativeEthernet Library**](https://github.com/vjmuzik/NativeEthernet) for **Teensy 4.1**
2. Add Version String.

khoih-prog, 2020-12-05:
Hi @jimmie11 

Just release [**EthernetWebServer_SSL v1.3.0**](https://github.com/khoih-prog/EthernetWebServer_SSL/releases/tag/v1.3.0)

Can you test to see if there is any more issue and give feedback.

Your contribution is noted in [**Contributions-and-Thanks**](https://github.com/khoih-prog/EthernetWebServer_SSL#contributions-and-thanks)

---

### Release v1.3.0

1. Add support to [**NativeEthernet Library**](https://github.com/vjmuzik/NativeEthernet) for **Teensy 4.1**

teatralnik, 2021-03-13:
Thanks to everyone for solving this isue!