khoih-prog/EthernetWebServer #16: NTP Server not working on teensy 4.1

jimmie11, 2020-12-05:
Thank you for your efforts in continuing to update the library.

I tried the NTP example.  The Teensy 4.1 gets an IP address but does not receive a response from the NTP server.

I tried changing the timeserver address to "129.6.15.28", but still no response.

khoih-prog, 2020-12-05:
Can you test it without using [**EthernetWebServer**](https://github.com/khoih-prog/EthernetWebServer) to be sure this is the issue of [**NativeEthernet Library**](https://github.com/vjmuzik/NativeEthernet).

I don't think this issue has anything to do with this Library.

jimmie11, 2020-12-05:
Sure, will try it now and report.

BTW, while the example webserver_NativeEthernet works, the Advanced example
does not.   In the Advanced example, the board does get an address but the
server never responds.

khoih-prog, 2020-12-05:
As I haven't worked on Teensy for a quite some time, and still has **no NativeEthernet to test on Teensy 4.1**, it'll take much longer time to figure out what wrong with the [**NativeEthernet Library**](https://github.com/vjmuzik/NativeEthernet) and why the [**AdvancedWebServer_NativeEthernet** example](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/AdvancedWebServer_NativeEthernet) is not working.

If you can simplify the sketch, by not using this [EthernetWebServer lib](https://github.com/khoih-prog/EthernetWebServer), to be sure this is the issue of **NativeEthernet Library**](https://github.com/vjmuzik/NativeEthernet) as I suspect. Then post an issue in Teensy forum.

I'll order some parts to enable NativeEthernet on my Teensy 4.1, but it'll take some time. You're on  your own now to help find out what's wrong.

jimmie11, 2020-12-05:
OK, thanks, will try and update you.
-----------------------------------------------------------------------------
I tried an NTP example with the NativeEthernet library and it is also NOT
working.  Apparently there is a bug ...

https://forum.pjrc.com/threads/64253-Teensy-4-1-UDP-and-web-server-drop-packets?p=258298

khoih-prog, 2020-12-05:
The [**NativeEthernet Library**](https://github.com/vjmuzik/NativeEthernet) is still new and buggy. Try our best to help make it better for everybody.

If urgent, I think you can try using the Ethernet, EthernetLarge, Ethernet2/3 options for the Teensy 4.1 W5x00

jimmie11, 2020-12-05:
The NTP example is working with the Teensy 4.1

I tried multiple NTP servers but the only one that worked was time.google.com.

I have attached an example the uses the NativeEthernet library and also works with "time.google.com"
[NTP.zip](https://github.com/khoih-prog/EthernetWebServer/files/5647800/NTP.zip)

closed 2020-12-05 by jimmie11

khoih-prog, 2020-12-05:
Can't download the NTP.zip as Chrome blocks complaining some virus, etc.
Can you post the code directly here.

khoih-prog, 2020-12-05:
It's possible the issue with [**AdvancedWebServer_NativeEthernet** example](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/AdvancedWebServer_NativeEthernet) has some relation with your recent issue [Teensy-4-1-Ethernet-Error-client-availableForWrite?](https://forum.pjrc.com/threads/64830-Teensy-4-1-Ethernet-Error-client-availableForWrite?highlight=NativeEthernet) and more.

khoih-prog, 2021-09-04:
Hi @jimmie11 

Could you please try again using [**QNEthernet** library](https://github.com/ssilverman/QNEthernet) ?

In case you haven't tried it, have a quick look at this very good [**New lwIP-based Ethernet library for Teensy 4.1**](https://forum.pjrc.com/threads/68066-New-lwIP-based-Ethernet-library-for-Teensy-4-1)

---

### Major Releases v1.6.0

1. Add support to [QNEthernet Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet
2. Update examples with new features