khoih-prog/EthernetWebServer #17: Help with the ESP32

ArminPP, 2020-12-17:
Hi there,
I have to admit that in using this library I am completely lost.

I have a m5STack with an ethernet unit (ESP32 & W5500) and cannot compile any of the demos.
Because I don't know how and where to define my hardware.

If there is a ready-made demo code for ESP32 and W5500 Ethernet Shield, I will be very happy.

Thanks,
Armin

khoih-prog, 2020-12-17:
Thanks for your interest in the library.

Because there are **too many different combinations of boards and Ethernet shields**, the examples, even only designed for most commonly used boards, are still very complex for deal with. I think that's **versatility and functionality** are better for all users.

The ESP32/ESP8266, with its build-in and powerful WiFi, seldom needs to be used with Ethernet. Even it's currently supported, you have to read/understand the example's code, then modify the example's code to enable it.

With your sincere request, I'll make a new simple example to deal with ESP32/ESP8266 and post it on the library within several days.

As I don't have the same `m5STack with an ethernet unit (ESP32 & W5500)`, I'll make it for generic ESP32+W5500. I'll appreciate if you can send any info about that board (pin-out, etc.) to save me some time to research.

khoih-prog, 2020-12-17:
Just added 2 examples for ESP32/ESP8266

1. [**WebClient_ESP**](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/WebClient_ESP)
2. [**WebClientRepeating_ESP**](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/WebClientRepeating_ESP)

Hereafter is the debug terminal output when running [**WebClientRepeating_ESP**](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/WebClientRepeating_ESP) on ESP32_DEV
```
Starting WebClientRepeating_ESP on ESP32_DEV with W5x00 using EthernetLarge Library
EthernetWebServer Version v1.2.0
[ETHERNET_WEBSERVER] =========== USE_ETHERNET_LARGE ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 23
[ETHERNET_WEBSERVER] MISO: 19
[ETHERNET_WEBSERVER] SCK: 18
[ETHERNET_WEBSERVER] SS: 5
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] ESP32 setCsPin: 22
_pinCS = 0
W5100 init, using SS_PIN_DEFAULT = 22, new ss_pin = 10, W5100Class::ss_pin = 22
W5100::init: W5500, SSIZE =8192
=========================
Currently Used SPI pinout:
MOSI:23
MISO:19
SCK:18
CS/SS:22
=========================
Using mac index = 2
Connected! IP address: 192.168.2.79

Starting connection to server...

Connecting...
HTTP/1.1 200 OK
Server: nginx/1.4.2
Date: Thu, 17 Dec 2020 19:04:59 GMT
Content-Type: text/plain
Content-Length: 2263
Last-Modified: Wed, 02 Oct 2013 13:46:47 GMT
Connection: close
Vary: Accept-Encoding
ETag: "524c23c7-8d7"
Accept-Ranges: bytes

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    
   ;;;;;;;;:`   `;;;;;;;;;     ,;;;;;;;;.`   .;;;;;;;;   
  .;;;;;;,         :;;;;;;;   .;;;;;;;          ;;;;;;;  
  ;;;;;;             ;;;;;;;  ;;;;;;,            ;;;;;;. 
 ,;;;;;               ;;;;;;.;;;;;;`              ;;;;;; 
 ;;;;;.                ;;;;;;;;;;;`      ```       ;;;;;`
 ;;;;;                  ;;;;;;;;;,       ;;;       .;;;;;
`;;;;:                  `;;;;;;;;        ;;;        ;;;;;
,;;;;`    `,,,,,,,,      ;;;;;;;      .,,;;;,,,     ;;;;;
:;;;;`    .;;;;;;;;       ;;;;;,      :;;;;;;;;     ;;;;;
:;;;;`    .;;;;;;;;      `;;;;;;      :;;;;;;;;     ;;;;;
.;;;;.                   ;;;;;;;.        ;;;        ;;;;;
 ;;;;;                  ;;;;;;;;;        ;;;        ;;;;;
 ;;;;;                 .;;;;;;;;;;       ;;;       ;;;;;,
 ;;;;;;               `;;;;;;;;;;;;                ;;;;; 
 `;;;;;,             .;;;;;; ;;;;;;;              ;;;;;; 
  ;;;;;;:           :;;;;;;.  ;;;;;;;            ;;;;;;  
   ;;;;;;;`       .;;;;;;;,    ;;;;;;;;        ;;;;;;;:  
    ;;;;;;;;;:,:;;;;;;;;;:      ;;;;;;;;;;:,;;;;;;;;;;   
    `;;;;;;;;;;;;;;;;;;;.        ;;;;;;;;;;;;;;;;;;;;    
      ;;;;;;;;;;;;;;;;;           :;;;;;;;;;;;;;;;;:     
       ,;;;;;;;;;;;;;,              ;;;;;;;;;;;;;;       
         .;;;;;;;;;`                  ,;;;;;;;;:         
                                                         
                                                         
                                                         
                                                         
    ;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   
```

Good Luck,

closed 2020-12-17 by khoih-prog

ArminPP, 2020-12-17:
Thank you very much for your kind support :-)

I was about to upload a snippet of code, but my computer is not working, so I write this via mobile phone.

Again, thank for your quick help, I will try the demo as soon as possible.
Greetings from Austria :-)

khoih-prog, 2020-12-17:
Just check the [**M5Stack W5500 Ethernet module**](https://m5stack.com/products/lan-module), the CS/SS pin of that module is **GPIO26**

[image]

So you have to change the example code to

```
#ifndef USE_THIS_SS_PIN
    //#define USE_THIS_SS_PIN   22    // For ESP32
    #define USE_THIS_SS_PIN   26      // For M5Stack W5500 Ethernet module.
#endif
```

Please post the test result / issue here to help other users.

ArminPP, 2020-12-17:
Yes, of course I will post my result here. 👍 
But please be patient, I have got some serious problems with my PC. 
I am really sorry :-/

khoih-prog, 2020-12-17:
I just updated the examples to better and easier for you to use

Just modify in the example to
 from

```
#define USING_M5STACK_W5500     false
``` 
to

```
#define USING_M5STACK_W5500     true
```
like this

```
#ifndef USE_THIS_SS_PIN
    #define USING_M5STACK_W5500     true
    #if USING_M5STACK_W5500
      #warning Using M5Stack_Core_ESP32 with W5500 mudule
      #define USE_THIS_SS_PIN   26    // For M5Stack_Core_ESP32 with W5500 mudule
    #else
      #define USE_THIS_SS_PIN   22    // For ESP32
    #endif
#endif
```

ArminPP, 2020-12-18:
Good Morning,
yesterday I had problems with the Ethernet.h library that were completely new to me ...

I have experience with Arduino UNO / MEGA and the Ethernet shield (and Ethernet.h Lib) and also the ESP32 gateway from Olymex (but with the old ETH.h library).
Yesterday I couldn't compile an Ethernet demo without errors, the message kept coming up was

`cannot declare variable 'server' to be of abstract type 'EthernetServer'`

I thought my system was corrupted, so I tried another computer today.

Same problem.

After some researching I found the following suggestion:
[(https://github.com/arduino-libraries/Ethernet/issues/88)]
[(https://github.com/adafruit/Ethernet2/issues/32)]
(and so on...)

So I changed the following:
```
in file -> C:\Users\armin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Server.h

class server: public print
{
public:
    // virtual void begin (uint16_t port = 0) = 0;
  virtual void begin () = 0;
};
```
Now I can compile programs with the Library Ethernet.h without errors, but I have a strange feeling that I have to "hack" a "core lib" for it ...

**Your current demos for the m5Stack are working perfectly - thank you again!** 

Have you had similar experiences with compiling the Ethernet.h lib with ESP32 cores?

khoih-prog, 2020-12-18:
If you read the README carefully, you'll see

[image]