khoih-prog/EthernetWebServer #18: Equivalent for "server.on"

jimmie11, 2020-12-25:
A very nice and complete webserver example has been posted by David Bird at:

[](https://github.com/G6EJD/ESP-SMART-Thermostat)

I am trying to convert that example to use your library but getting a compile error for 

`server.on`

Is there an equivalent to that statement?

Thank you.

khoih-prog, 2020-12-25:
Those examples are written for ESP32/ESP8266 using **ESPAsyncWebServer** which is not supported by the [**EthernetWebServer Library**](https://github.com/khoih-prog/EthernetWebServer).

You can convert the example to use synchronous (ESP8266)WebServer, then you can use them with this library. The syntax is very similar, such as 

```
server.on("/", handleSomething);
```

Good Luck,

closed 2020-12-25 by khoih-prog

jimmie11, 2020-12-25:
Thank you but Server.on is what causes the compile error.  How will using the same statement and changing the library to synchronous help?

khoih-prog, 2020-12-25:
You certainly have to first rewrite the examples to use (ESP8266)WebServer. That will modify the function call from asynch way

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) 
```

to synchronous way

```
server.on("/", handleSomething);
```
then port he modified examples to EthenetWebServer.

It's better to spend some more time to research the differences between Async and Sync examples.

khoih-prog, 2020-12-25:
This is the [**Async_HelloServer**](https://github.com/khoih-prog/AsyncWebServer_STM32/blob/master/examples/Async_HelloServer/Async_HelloServer.ino) example written for [**AsyncWebServer_STM32 library**](https://github.com/khoih-prog/AsyncWebServer_STM32)

```
void setup(void)
{
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);

  Serial.begin(115200);
  Serial.println("\nStart Async_HelloServer on " + String(BOARD_NAME));
...

  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request)
  {
    handleRoot(request);
  });

  server.on("/inline", [](AsyncWebServerRequest * request)
  {
    request->send(200, "text/plain", "This works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();

  Serial.print(F("HTTP EthernetWebServer is @ IP : "));
  Serial.println(Ethernet.localIP());
}

void loop(void)
{
}
```

and the similar [**HelloServer**](https://github.com/khoih-prog/EthernetWebServer/blob/master/examples/HelloServer/HelloServer.ino) written for sync [**EthernetWebServer library**](https://github.com/khoih-prog/EthernetWebServer)

```
void setup(void)
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial);

  Serial.print("\nStarting HelloServer on " + String(BOARD_TYPE));
  Serial.println(" with " + String(SHIELD_TYPE));
  Serial.println("EthernetWebServer Version " + String(ETHERNET_WEBSERVER_VERSION));
...

  server.on(F("/"), handleRoot);

  server.on(F("/inline"), []() 
  {
    server.send(200, F("text/plain"), F("This works as well"));
  });

  server.onNotFound(handleNotFound);

  server.begin();

  Serial.print(F("HTTP EthernetWebServer is @ IP : "));
  Serial.println(Ethernet.localIP());
}

void loop(void)
{
  server.handleClient();
}
```