khoih-prog/EthernetWebServer #24: Support WT32-ETH01  (ESP32 / LAN8720)

BorgMcz, 2021-02-02:
Good day,

I would like to ask if it would be possible to add support for the following board. 
[https://www.makeuseof.com/this-6-esp32-board-is-a-diy-smart-home-dream/](https://www.makeuseof.com/this-6-esp32-board-is-a-diy-smart-home-dream/)
[https://www.seeedstudio.com/Ethernet-module-based-on-ESP32-series-WT32-ETH01-p-4736.html](https://www.seeedstudio.com/Ethernet-module-based-on-ESP32-series-WT32-ETH01-p-4736.html)
[https://github.com/Aircoookie/WLED/issues/1027#issuecomment-750540512](https://github.com/Aircoookie/WLED/issues/1027#issuecomment-750540512)

It is a very compact and cheap design. PHY support is implemented directly into the ESP32 SDK.

khoih-prog, 2021-02-03:
Thank for your interest in this library.

As you know, the WT32-ETH01 (ESP32 / LAN8720) is using special and not fully compatible AT-commands to control ESP32-WiFi and Ethernet. This is very different from what is supported by this [**EthernetWebServer library**](https://github.com/khoih-prog/EthernetWebServer). Therefore, **this library will never support that type of shield.**

The closest libraries you can use for WiFi, hopefully working OK, are

1. [**ESP8266_AT_WebServer**](https://github.com/khoih-prog/ESP8266_AT_WebServer)
2. [**ESP_AT_WiFiManager**](https://github.com/khoih-prog/ESP_AT_WiFiManager)
3. [**ESP_AT_Lib**](https://github.com/khoih-prog/ESP_AT_Lib)

I'll try to ask SeeedStudio to send or buy some samples of this interesting WiFi+Ethernet shield to see if I can **write a special library (similar to those AT-command libraries) for them**. But I can't promise anything as I'm so busy right now, just in the long wish list.

closed 2021-02-03 by khoih-prog

BorgMcz, 2021-02-09:
Good day,

The use of this PHY chip is directly supported by the ESP and its SDK library. A simple example of activation can be viewed here.

I ordered two pieces so I could try, now I'm waiting for delivery. This board has great potential.

[https://www.letscontrolit.com/forum/viewtopic.php?p=49844#p49844](https://www.letscontrolit.com/forum/viewtopic.php?p=49844#p49844)

BorgMcz, 2021-02-13:
I already have the module in my hand. I managed basic operations such as sending an email and the like. Now I would like to use your interface to work with a web server.
Here I attach the tested settings for this hardware.

`/* 
   * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
   * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN		
#define ETH_POWER_PIN   16						                // Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_TYPE        ETH_PHY_LAN8720			                     // Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_ADDR        1						                         // I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_MDC_PIN     23						                     // Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDIO_PIN    18						                  // Pin# of the I²C IO signal for the Ethernet PHY`

ldijkman, 2021-03-28:
maybe for someone some help 
NOT another century AT wired phone modem commands ;-)
but arduino ide examles for WT32-ETH01
https://github.com/ldijkman/WT32-ETH01-LAN-8720-RJ45-

khoih-prog, 2021-03-28:
@ldijkman 

Thanks for your repo which will help many users.

I'll definitely add the **EthernetWebServer** support to this interesting WT32-ETH01 board when I receive (still somewhere)

Regards,

khoih-prog, 2021-07-06:
Hi all,

FYI, the support for good **WT32-ETH01** board has been provided by the new library [**WebServer_WT32_ETH01**](https://github.com/khoih-prog/WebServer_WT32_ETH01), which is just released today.

Cheers,

khoih-prog, 2021-07-06:
Hi @ldijkman 

Thanks for you good library which helps and saves me a lot of time to start with the new **WT32-ETH01** board.

Best Regards,

SurfGargano, 2021-11-22:
In the file webServer_WT32_ETH01_Impl.h ther eare some cases like 
case ARDUINO_EVENT_ETH_START:
When I start the compilation some errors are display that ARDUINO_EVENT_ETH_START (and other) not defined.
Has I include some other files ?

khoih-prog, 2021-11-22:
Please put the issue in the correct library [**WebServer_WT32_ETH01**](https://github.com/khoih-prog/WebServer_WT32_ETH01) next time.

---

PIO is using ESP32 core v1.0.6 now, the current library version is using v2.0.0+ by default.

You have to either

1. Using Arduino IDE and ESP32 core v2.0.1
2. Using PIO and library version v1.2.0-. Read [**Important notes**](https://github.com/khoih-prog/WebServer_WT32_ETH01#important-notes)
3. If using library version v1.3.0, add in your code,

```
// Uncomment to use ESP32 core v1.0.6-
#define USING_CORE_ESP32_CORE_V200_PLUS     false
```

ldijkman, 2021-11-22:
> Hi @ldijkman
> 
> Thanks for you good library which helps and saves me a lot of time to start with the new **WT32-ETH01** board.
> 
> Best Regards,

Haha i think i like your espasync wifimanager
For getting me started
Easy home automation mdns logical hostname
Http://living.local etc with auto linked list of devices in network
https://github.com/ldijkman/Hey_Electra