khoih-prog/EthernetWebServer #25: Unable to Compile Any EthernetWebServer Examples on ESP32-Wrover-E

HWEDK, 2021-02-12:
I have tried multiple items at this point to include reinstalls and different machines on Windows 10. I have verified the prereqs per the instructions here--I was working offline, so I also checked on an online machine with the latest direct downloads for dependencies etc. I should note that I used the following ESP32 ethernet example and verified that my W5500 board and my ESP32 Devkitc_v4 board with wrover-E chip are functional and connecting properly otherwise--https://github.com/PuceBaboon/ESP32_W5500_NTP_CLIENT. 

Finally, I tried variations of the fix for the Server.h file and made sure that the ethernet.h, ethernet.cpp, and ethernetserver.cpp are the correct fixes per recommendations. 

I get the below-pasted set of errors when I try to compile, and it is the same in all variations on both machines and in different versions. It seems to be a type mismatch or something. I have the board manager set at 115200 baud speed, correct port, ESP32 Wrover Module board selected, and default 4MB partition scheme with SPIFFS.

I am just beginning to approach code professionally, so assistance on this for a poor former Controls Engineer would be greatly appreciated. I really need this to work. I am trying to flesh out an ethernet connection that can connect to a basic HTML/HTTP website for external control, (no PoE won't work here, and no Wi-Fi is not an option), so I'd take recommendations as well for alternate paths to make something exactly like this work but for ethernet https://randomnerdtutorials.com/esp32-access-point-ap-web-server/. 

In file included from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_839864\WebServer.ino:15:0:
sketch\defines.h:235:4: warning: #warning Use ESP32 architecture [-Wcpp]
   #warning Use ESP32 architecture
    ^
In file included from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_839864\WebServer.ino:15:0:
sketch\defines.h:329:8: warning: #warning Using Ethernet lib [-Wcpp]
       #warning Using Ethernet lib
        ^
In file included from sketch\defines.h:344:0,
                 from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_839864\WebServer.ino:15:
C:\Users\admin\Documents\Arduino\libraries\EthernetWebServer\src/EthernetWebServer.h:94:4: warning: #warning Using Ethernet library from EthernetWebServer [-Wcpp]
   #warning Using Ethernet library from EthernetWebServer
    ^
In file included from sketch\defines.h:344:0,
                 from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_839864\WebServer.ino:15:
C:\Users\admin\Documents\Arduino\libraries\EthernetWebServer\src/EthernetWebServer.h:109:4: warning: #warning SENDCONTENT_P_BUFFER_SZ using default 4 Kbytes [-Wcpp]
   #warning SENDCONTENT_P_BUFFER_SZ using default 4 Kbytes
    ^
In file included from sketch\defines.h:344:0,
                 from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_839864\WebServer.ino:15:
C:\Users\admin\Documents\Arduino\libraries\EthernetWebServer\src/EthernetWebServer.h:326:21: error: cannot declare field 'EthernetWebServer::_server' to be of abstract type 'EthernetServer'
     EthernetServer  _server;
                     ^
In file included from sketch\defines.h:328:0,
                 from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_839864\WebServer.ino:15:
C:\Program Files (x86)\Arduino\libraries\Ethernet\src/Ethernet.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^
In file included from C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:152:0,
                 from sketch\WebServer.ino.cpp:1:
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;
                  ^
WebServer:19:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'
 EthernetServer server(80);
                ^
exit status 1
cannot declare variable 'server' to be of abstract type 'EthernetServer'

khoih-prog, 2021-02-12:
Hi @HWEDK 

Thanks for your interest in the library. I used to be Controls Engineer at some time, and understand how you'd love the jobs for your whole life.

From your error message, it seems that you forgot to copy the `Server.h` to the correct location, in your case

```
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Server.h
``` 

That'll fix the issue you mentioned

```

#ifndef server_h
#define server_h

#include "Print.h"

class Server: public Print
{
public:
    // KH, change to fix compiler error for EthernetWebServer
    // error: cannot declare field 'EthernetWebServer::_server' to be of abstract type 'EthernetServer'
    // virtual void begin(uint16_t port=0) =0;
    //virtual void begin() = 0;
    void begin() {};
};

#endif
```

closed 2021-02-12 by khoih-prog

HWEDK, 2021-02-13:
Thanks a lot, very fast response too. At least I understand the code better after digging around so much...
If I may, another dumb item I must have overlooked, I get the following on serial terminal with 0.0.0.0 IP?
Is this something with my own local network. I set the defines.h IP to the range of my computer.  
Something else in the config I'm missing? 

Starting WebServer on ESP32 with W5x00 using Ethernet Library
EthernetWebServer v1.2.1
[ETHERNET_WEBSERVER] =========== USE_ETHERNET ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 23
[ETHERNET_WEBSERVER] MISO: 19
[ETHERNET_WEBSERVER] SCK: 18
[ETHERNET_WEBSERVER] SS: 5
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] ESP32 setCsPin: 22
=========================
Currently Used SPI pinout:
MOSI:23
MISO:19
SCK:18
SS:5
=========================
Using mac index = 0
EthernetWebServer started @ IP address: 0.0.0.0

khoih-prog, 2021-02-13:
It's possible that you connect the W5500 CS/SS pin to the **wrong pin** from what specified in the sketch..

According to your current (default) settings

```
[ETHERNET_WEBSERVER] ESP32 setCsPin: 22
```
the W5500 CS/SS pin must be connected to GPIO22 of ESP32.

It's possible that you're connect it to GPIO5 (many cases, such as in [**ESP32_W5500_NTP_CLIENT**](https://github.com/PuceBaboon/ESP32_W5500_NTP_CLIENT)

If so, you can just modify [**WebServer.ino#L109-L111**](https://github.com/khoih-prog/EthernetWebServer/blob/master/examples/WebServer/WebServer.ino#L109-L111) to change the pin used for CS/SS

From pin **GPIO22**

```
#ifndef USE_THIS_SS_PIN
  #define USE_THIS_SS_PIN   22    // For ESP32
#endif
```

to pin **GPIO5** (or whatever pin you're using)

```
#ifndef USE_THIS_SS_PIN
  #define USE_THIS_SS_PIN   5    // For ESP32, or whatever GPIO pin you're using
#endif
```

Good Luck,

You can also try to use DHCP IP by keeping [**WebServer.ino#L185-L187**](https://github.com/khoih-prog/EthernetWebServer/blob/master/examples/WebServer/WebServer.ino#L185-L187) as is

```
// Use Static IP
//Ethernet.begin(mac[index], ip);
Ethernet.begin(mac[index]);
```

HWEDK, 2021-02-15:
YES! I got it...Your line about DHCP made me realize I crossed code that was trying to expect both DHCP and Static. Also, yes, needed the line with reset being 22, and ensuring CS are set to correct pin in code also needed doing. I have the HTML displaying now. 

Now I just have to merge WiFi, Ethernet, Filesystem, and SPI handling

khoih-prog, 2021-02-15:
Good to know you're making good progress by yourself. The acquired experience will definitely follow you forever.