khoih-prog/EthernetWebServer #27: Feature Request:  Add a RequestHandler subclass which passes the server to the function

EmperorArthur, 2021-03-03:
Hello,

I am currently examining the feasibility of switching from our current custom (but extremely fast) Arduino web server code.
In doing so, I noticed that the default on methods, and `FunctionRequestHandler` only call functions which have no arguments.

That is a serious issue, since it means the server code must be a global, and breaks all sorts of best practices.

My cheap and dirty work around is something like this, but due to the next bug, I can't even test it with our current code base.

```
typedef vl::Func<void(EthernetWebServer&)> THandlerFunctionWithServer;
class FunctionRequestHandlerWithServer: public FunctionRequestHandler {
public:
    FunctionRequestHandlerWithServer(THandlerFunctionWithServer fn,
                                     THandlerFunctionWithServer ufn, const String &uri, HTTPMethod method)
    : FunctionRequestHandler([](){}, [](){}, uri, method), _fns(fn), _ufns(ufn)
    {
    }

    bool handle(EthernetWebServer& server, HTTPMethod requestMethod, String requestUri) override
    {
        ETW_UNUSED(server);

        if (!canHandle(requestMethod, requestUri))
            return false;

        _fns(server);
        return true;
    }

    void upload(EthernetWebServer& server, String requestUri, HTTPUpload& upload) override
    {
        ETW_UNUSED(server);
        ETW_UNUSED(upload);

        if (canUpload(requestUri))
            _ufns(server);
    }

protected:
    THandlerFunctionWithServer _fns;
    THandlerFunctionWithServer _ufns;
};

void handlerFunction(EthernetWebServer& server) {
    server.send(404, "application/json", "Something";
}
```

khoih-prog, 2021-03-03:
HI,

Thanks for the **Feature Request**.

To save both of us spending some time back and forth, could you please provide some [**MRE** : How to create a Minimal, Reproducible Example](https://stackoverflow.com/help/minimal-reproducible-example) to demonstrate your case. I'll certainly spend some time later to investigate and fix the library.

Regards,

EmperorArthur, 2021-03-03:
Hello,

Unfortunately, this request is only truly needed when #28 is implemented.  Our use case is the server/router living in "main.cpp", and each individual component (".h/.cpp" file) having their own `THandlerFunction`.  This allows for modularity of the system, and is how almost all Model View Controller (MVC) web frameworks operate.\*

As #28 is closed as WontFix, this issue is mute for our use case.

\*  For more information, Dask and Laravel are my recommended starting points, depending on language preference.

khoih-prog, 2021-03-03:
Thanks for your interests and feedbacks anyway.

I'll spend some time to add separate cpp folder later, but have to close the issue now.

closed 2021-03-03 by khoih-prog