khoih-prog/EthernetWebServer #28: Error on PlatformIO "multiple definition of `EthernetWebServer::.."

EmperorArthur, 2021-03-03:
Hello,

Due to the way the code is written, `#include <EthernetWebServer.h>` can never be used in a header file.

It always results in a multiple definition error.

I believe the problem can be traced to these two lines in "EthernetWebServer.h"

```
#include "EthernetWebServer-impl.h"
#include "Parsing-impl.h"
```

It may be better to convert the "*-impl.h" files to instead be ".cpp" files, which are linked at compile time.

khoih-prog, 2021-03-03:
Hi,

Thanks for your interests in the library and your [**Feature Request:** Add a RequestHandler subclass which passes the server to the function #27](https://github.com/khoih-prog/EthernetWebServer/issues/27)

For more info on why and how to use correctly or fix this, check 

1. [HOWTO Fix Multiple Definitions Linker Error](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#howto-fix-multiple-definitions-linker-error)
2. [Fixed linker errors when included in multiple .cpp files #1](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/pull/1)
3.  and many similar posts here and there

I'll consider to add cpp version besides the h-only version later if there are more requests as this issue happens only in rare use-cases or when inclusion is not carefully considered.

closed 2021-03-03 by khoih-prog

khoih-prog, 2021-03-03:
More notes

> Due to the way the code is written, #include <EthernetWebServer.h> can never be used in a header file.

I don't know if you go through the examples before writing this line

For examples [#include <EthernetWebServer.h> @ line 344 of **defines.h**](https://github.com/khoih-prog/EthernetWebServer/blob/master/examples/AdvancedWebServer/defines.h#L344)

EmperorArthur, 2021-03-03:
Hello,

Apologies for not being more clear.  I meant to say, "`#include <EthernetWebServer.h>` can never be included in a header file which is included by more than one ".cpp" file."

We were unaware of the "AsyncHTTPRequest_Generic library" until now, and thank you for responding to our question.
I had never even heard of your .  I see your response to my suggestion there.

Unfortunately, a requirement of the project is that someone can download the code and follow the standard PlatformIO build instructions to run it, therefore modifying the library files is nor allowed.  In addition, as this is a relatively large code base, eschewing C/C++ convention and converting everything to "-impl.h" files is also a hard pass.

With this in mind, we have determined that your library is not suitable for a production environment, and will continue searching for alternatives.

Thank you for your time.