khoih-prog/EthernetWebServer #29: MDNS support

pptsk, 2021-03-21:
Hi,
Does this library support MDNS for esp32?
Arduino for esp32 has ESPmDNS library but I cant find any example here...
Thank you

khoih-prog, 2021-03-21:
It's currently not supporting mDNS, which is still in the [**TODO list**](https://github.com/khoih-prog/EthernetWebServer#to-do)

You can try to use `ESPmDNS` library in ESP32, provided you still have WiFi working besides Ethernet because `ESPmDNS` is relying on WiFi.

In the mean time, try an independent mDNS library such as [**ArduinoMDNS**](https://github.com/arduino-libraries/ArduinoMDNS).

As I don't have time to work on this TODO list now, I have to close the issue and it as `enhancement request`, hopefully to fix it in the near future.

closed 2021-03-21 by khoih-prog

khoih-prog, 2021-03-21:
You can try these [**MDNS_Generic examples**](https://github.com/khoih-prog/MDNS_Generic) written for Ethernet  and can be merged with [**EthernetWebServer** library](https://github.com/khoih-prog/EthernetWebServer)

1. [ResolvingHostNames](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/Ethernet/ResolvingHostNames)
2. [DiscoveringServices](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/Ethernet/DiscoveringServices)
3. [RegisteringServices](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/Ethernet/RegisteringServices)
4. [RegisteringServicesWithTxtRecord](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/Ethernet/RegisteringServicesWithTxtRecord)