khoih-prog/EthernetWebServer #31: FNET dns-client (on Teensy 4.1) doesn't resolve hostnames of Azure Table Storage accounts

RoSchmi, 2021-05-07:
This is not an issue of this library but of the FNET library used by the NativeEthernet library on Teensy 4.1.
I'm just searching for people to take part in the discussion whether this is an issue of the FNET library or not.

https://github.com/butok/FNET/issues/7

https://stackoverflow.com/questions/67421802/should-rfc-conform-dns-clients-resolve-urls-of-azure-table-storage-accounts-whic

On the first I would be happy if someone could confirm that on the Teensy 4.1 FNET doesn't resolve the Ip addresses of CNAME records.
Kind regards
RoSchmi

khoih-prog, 2021-05-07:
@RoSchmi 

Thanks for letting us know about the issue. I'm afraid I can't spend time on this issue, but definitely will follow up.

I'm closing the issue now as it's not from this library. Keep us posted anyway.

Regards,

closed 2021-05-07 by khoih-prog

RoSchmi, 2021-05-08:
Thanks anyway, everbody has to set their priorities.

khoih-prog, 2021-09-04:
Hi @RoSchmi 

Could you please try again using [**QNEthernet** library](https://github.com/ssilverman/QNEthernet) ?

In case you haven't tried it, have a quick look at this very good [**New lwIP-based Ethernet library for Teensy 4.1**](https://forum.pjrc.com/threads/68066-New-lwIP-based-Ethernet-library-for-Teensy-4-1)

---

### Major Releases v1.6.0

1. Add support to [QNEthernet Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet
2. Update examples with new features

RoSchmi, 2021-09-06:
Thanks @khoih-prog,
in the meantime this issue of the FNET library  has been fixed in the vjmuzik/FNET fork.

Today I tried to run my application with the QNEthernet library and it seems to work flawlessly. There are no problems resolving the Azure Storage url.
https://github.com/RoSchmi/AzureDataSender_Teensy_QnEthernet

khoih-prog, 2021-09-06:
Thanks and Regards @RoSchmi 

You're so fast to release that new repo. to support **QNEthernet**.

I was impressed with the robustness and stability of the new lwip-based QNEthernet, and decided to port many libraries to use it. 

Check the status at [New and Modified Libraries to support Teensy 4.1 QNEthernet](https://forum.arduino.cc/t/new-and-modified-libraries-to-support-teensy-4-1-qnethernet/902354)

I believe you also can help promote this new QNEthernet library so that the lib's author [Shawn Silverman](https://github.com/ssilverman) will make many more improvements and other users can have this alternative.