khoih-prog/EthernetWebServer #33: WebServer not functioning on Teensy 4.1

MoosKarneval181, 2021-05-21:
### Describe the bug

When the advanced functionality of this library is used on a Teensy 4.1, requests are never handled and time out.
It also stops going trough the loop until the request is cancelled.

### Steps to Reproduce

1. Setup EthernetWebServer with advanced functions, like in [AWS-example](https://github.com/khoih-prog/EthernetWebServer/blob/master/examples/AdvancedWebServer/AdvancedWebServer.ino)
2. Run it on a Teensy 4.1
3. Call the webpage

### Expected behavior

Teensy sending "Hello World" as a response.

### Actual behavior

The request is never handled by Teensy.

### Serial Output

```
10.0.0.3
..........
```

### Code used for testing

```
#include <EthernetWebServer.h>
#include <NativeEthernet.h>

#define _ETHERNET_WEBSERVER_LOGLEVEL_ 4
#define USE_NATIVE_ETHERNET true
#define USE_THIS_SS_PIN 10

byte mac[6] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

IPAddress ip(10, 0, 0, 3);
IPAddress gateway(10, 0, 0, 1);
IPAddress submask(255, 255, 255, 0);

EthernetWebServer server(80);

void setup()
{
    Serial.begin(115200);
    while (!Serial)
        ;

    Ethernet.init(USE_THIS_SS_PIN);
    Ethernet.begin(mac, ip, gateway, submask);

    Serial.println(Ethernet.localIP());

    server.on("/", []()
              { server.send(200, "text/html", "Hello World"); });

    server.onNotFound([]()
                      { server.send(404, "Content not found"); });

    server.begin();
}
int lastBeat = 0;
void loop()
{
    if (millis() - lastBeat >= 5000)
    {
        Serial.print(".");
        lastBeat = millis();
    }
    server.handleClient();
}
```

khoih-prog, 2021-05-21:
Hi

Thanks for using the library.

I suggest you start with the working and more complex Server examples as follows, I just tested OK and running reliably. Then modify then according to your requirements.

We're sorry We certainly don't have time to spend on your issue, unless you use those examples and prove they're not working.

1. [AdvancedWebServer](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/AdvancedWebServer)

```

Starting AdvancedWebServer on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
EthernetWebServer v1.5.0
[EWS] ======== USE_NATIVE_ETHERNET ========
[EWS] Default SPI pinout:
[EWS] MOSI: 11
[EWS] MISO: 12
[EWS] SCK: 13
[EWS] SS: 10
[EWS] =========================
[EWS] Board : TEENSY 4.1 , setCsPin: 10
=========================
Currently Used SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
=========================
Using mac index = 14
Connected! IP address: 192.168.2.97
HTTP EthernetWebServer is @ IP : 192.168.2.97
```

[image]

---

2. [WebServer_NativeEthernet](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/WebServer_NativeEthernet)

[image]

---

closed 2021-05-21 by khoih-prog

m0r13, 2021-12-02:
Hey,

I can confirm the above issue on a Teensy 4.1 using the AdvancedWebserverExample.
The Teensy does not accept any connections but is running / prints the heartbeats.
As a difference to @MoosKarneval181, `Ethernet.localIP()` is always `0.0.0.0`, whether static IP or DHCP. 
I'm using Arduino 1.8.16, Teensyduino 1.55, and I have applied the patches necessary for the library.

Interestingly, it works on a different PC with Arduino 1.8.15 and Teensyduino 1.54.
I have already tried a fresh install of Arduino 1.8.16 and patching only Stream.h manually (in case that the patches are out-of-date...).
Please let me know how we can debug this.

I would actually like to use the library with platformio but there I run into the same issue.
I hope that finding out what the issue is will help fixing it there as well.

Best regards
Moritz

khoih-prog, 2021-12-02:
I suggest you first trying the better [**QNEthernet**](https://github.com/ssilverman/QNEthernet) instead of [**NativeEthernet**](https://github.com/vjmuzik/NativeEthernet) library.

I've just tested and added support to [**UPnP_Generic**](https://github.com/khoih-prog/UPnP_Generic) for Teeensy 4.1 + QNEthernet several hours ago.

The result in [T41_QNEthernet_SimpleServer on TEENSY 4.1](https://github.com/khoih-prog/UPnP_Generic#15-t41_qnethernet_simpleserver-on-teensy-41)

> Interestingly, it works on a different PC with Arduino 1.8.15 and Teensyduino 1.54.

I'm afraid only you can find out what is the issue, such as installation, missing files, etc.

Good Luck,

m0r13, 2021-12-03:
Thanks, it works with QNEthernet. With platformio as well... must be some issue with the current Arduino distribution or NativeEthernet then.