khoih-prog/EthernetWebServer #38: Consecutive page refreshes finally leads to data corruption and Connection Refused

aagiannidis, 2021-11-06:
Dear all, first of all i need to congratulate you on this wonderful library. I dont want to waste anyones time but I am desperate to find a solution. I can see that this topic is somehow similar to a previous one but I am confident that it is a valid case to be examined.

I am using an arduino Mega with an ethernet shield 2. The device is powered from USB cable directly connected to laptop. I have developed my own program and came across some issues that I couldnt explain. Finally i went back to the drawing board and conducted my tests with the Simple Authentication example as provided. I could replicate the issues with this example so it makes me think there is a bug somewhere in the library.

The bug is as follows: after consecutive refreshes the arduino starts sending less and less data and finally i get the message connection refused on the browser. The arduino seems to be stuck as it produces nothing. However it responds to pings Ok.

Here is some info on how to replicate the bug:

Arduino 1.8.13 
Arduino Mega plus Ethernet Shield 2 featuring a W5500.
Example: Simple Authentication
Page to refresh: $IP ADDRESS$/login
Browser: Chrome and Firefox
Configuration settings: using either Ethernet or Ethernet2 library

Load the page and do about 200 refreshes. Refreshes should be done once the page is loaded. On every refresh the arduino should send 350 characters with the content

`<html><body><form action='/login' method='POST'>To log in, please use : admin/password<br>User:<input type='text' name='USERNAME' placeholder='user name'><br>Password:<input type='password' name='PASSWORD' placeholder='password'><br><input type='submit' name='SUBMIT' value='Submit'></form><br>You also can go <a href='/inline'>here</a></body></html>`

After 150-200 refreshes the arduino starts sending less and less characters. Please see provided log further down.

Also, if the refreshes are faster the arduino will also report corrupted data on its parsed received headers. Please see provided log further down.

Here is a perfect page refresh log that sends the full 350 character long web page:

```
13:03:10.270 -> [EWS] handleClient: New Client
13:03:10.304 -> [EWS] method:  GET
13:03:10.304 -> [EWS] url:  /login
13:03:10.304 -> [EWS] search:  
13:03:10.304 -> [EWS] headerName: Host
13:03:10.304 -> [EWS] headerValue: 192.168.1.30
13:03:10.304 -> [EWS] headerName: Connection
13:03:10.304 -> [EWS] headerValue: keep-alive
13:03:10.304 -> [EWS] headerName: Upgrade-Insecure-Requests
13:03:10.338 -> [EWS] headerValue: 1
13:03:10.372 -> [EWS] headerName: User-Agent
13:03:10.372 -> [EWS] headerValue: Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/95.0.4638.69 Safari/537.36
13:03:10.406 -> [EWS] headerName: Accept
13:03:10.406 -> [EWS] headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
13:03:10.440 -> [EWS] headerName: Accept-Encoding
13:03:10.440 -> [EWS] headerValue: gzip, deflate
13:03:10.440 -> [EWS] headerName: Accept-Language
13:03:10.474 -> [EWS] headerValue: en-US,en;q=0.9,el;q=0.8
13:03:10.474 -> [EWS] headerName: Cookie
13:03:10.474 -> [EWS] headerValue: ESPSESSIONID=0
13:03:10.474 -> [EWS] args:  
13:03:10.474 -> [EWS] args count:  0
13:03:10.474 -> [EWS] args:  
13:03:10.474 -> [EWS] args count:  0
13:03:10.474 -> [EWS] Request: /login
13:03:10.474 -> [EWS] Arguments: 
13:03:10.474 -> [EWS] Final list of key/value pairs:
13:03:10.474 -> [EWS] _handleRequest handle
13:03:10.474 -> Found cookie: ESPSESSIONID=0
13:03:10.474 -> [EWS] send1: len =  350
13:03:10.507 -> [EWS] content =  <html><body><form action='/login' method='POST'>To log in, please use : admin/password<br>User:<input type='text' name='USERNAME' placeholder='user name'><br>Password:<input type='password' name='PASSWORD' placeholder='password'><br><input type='submit' name='SUBMIT' value='Submit'></form><br>You also can go <a href='/inline'>here</a></body></html>
13:03:10.541 -> [EWS] _prepareHeader sendHeader Conn close
13:03:10.541 -> [EWS] send1: write header =  HTTP/1.1 200 OK
13:03:10.541 -> Content-Type: text/html
13:03:10.541 -> Content-Length: 350
13:03:10.541 -> Connection: close
13:03:10.541 -> 
13:03:10.541 -> 
13:03:10.541 -> [EWS] sendContent: Client.write content:  <html><body><form action='/login' method='POST'>To log in, please use : admin/password<br>User:<input type='text' name='USERNAME' placeholder='user name'><br>Password:<input type='password' name='PASSWORD' placeholder='password'><br><input type='submit' name='SUBMIT' value='Submit'></form><br>You also can go <a href='/inline'>here</a></body></html>
13:03:10.576 -> [EWS] _handleRequest OK
13:03:10.576 -> [EWS] handleClient: Don't keepCurrentClient
```
After 150-200 refreshes it sends less data. In this case it sends 237 characters only.

[... 282 lines cut ...]
22:09:28.418 -> [EWS] _parseRequest: Invalid request:  Cookie
22:09:28.418 -> [EWS] handleClient: Don't keepCurrentClient
22:09:28.452 -> [EWS] handleClient: New Client
22:09:28.452 -> [EWS] _parseRequest: Invalid request:  
22:09:28.452 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.143 -> [EWS] handleClient: New Client
22:09:32.143 -> [EWS] _parseRequest: Invalid request:  GET /l
22:09:32.143 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.143 -> [EWS] handleClient: New Client
22:09:32.143 -> [EWS] _parseRequest: Invalid request:  Host: 
22:09:32.177 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.177 -> [EWS] handleClient: New Client
22:09:32.177 -> [EWS] _parseRequest: Invalid request:  Connec
22:09:32.177 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.177 -> [EWS] handleClient: New Client
22:09:32.177 -> [EWS] _parseRequest: Invalid request:  Cache-
22:09:32.177 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.177 -> [EWS] handleClient: New Client
22:09:32.211 -> [EWS] _parseRequest: Invalid request:  Upgrad
22:09:32.211 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.211 -> [EWS] handleClient: New Client
22:09:32.245 -> [EWS] _parseRequest: Invalid request:  User-A
22:09:32.245 -> [EWS] handleClient: Don't keepCurrentClient
22:09:32.245 -> [EWS] handleClient: New Client
```

aagiannidis, 2021-11-06:
Just before the arduino becomes non-responsive i can see from the logs that prepareHeader is executing but the send function is not executed.

```
16:54:55.123 -> [EWS] handleClient: New Client
16:54:55.157 -> [EWS] method:  GET
16:54:55.157 -> [EWS] url:  /login
16:54:55.157 -> [EWS] search:  
16:54:55.157 -> [EWS] headerName: Host
16:54:55.157 -> [EWS] headerValue: 192.168.1.30
16:54:55.157 -> [EWS] headerName: Connection
16:54:55.157 -> [EWS] headerValue: keep-alive
16:54:55.157 -> [EWS] headerName: Cache-Control
16:54:55.157 -> [EWS] headerValue: max-age=0
16:54:55.190 -> [EWS] args:  
16:54:55.190 -> [EWS] args count:  0
16:54:55.190 -> [EWS] args:  
16:54:55.190 -> [EWS] args count:  0
16:54:55.190 -> [EWS] Request: /login
16:54:55.190 -> [EWS] Arguments: 
16:54:55.190 -> [EWS] Final list of key/value pairs:
16:54:55.190 -> [EWS] _handleRequest handle
16:54:55.190 -> [EWS] send1: len =  162
16:54:55.190 -> [EWS] content =  <html><body><form action='/login' method='POST'>To log in, please use : admin/password<br>User:<input type='text' name='USERNAME' placeholder='user name'><br><br
16:54:55.224 -> [EWS] _prepareHeader sendHeader Conn close
```

This bit is missing from the log (taken from full ok example above)

```
22:08:19.554 -> [EWS] send1: write header =   
22:08:19.554 -> 
22:08:19.554 -> [EWS] sendContent: Client.write content:  User:<input type='text' name='USERNAME' placeholder='user name'><br>
22:08:19.554 -> [EWS] _handleRequest OK
22:08:19.589 -> [EWS] handleClient: Don't keepCurrentClient
```

aagiannidis, 2021-11-06:
Also a malformed log before Chrome says Err Connection Refused.
Note how the header values are malformed.

```
17:12:00.600 -> [EWS] handleClient: New Client
17:12:00.600 -> [EWS] method:  GET
17:12:00.600 -> [EWS] url:  /login
17:12:00.600 -> [EWS] search:  
17:12:00.600 -> [EWS] headerName: Host
17:12:00.600 -> [EWS] headerValue: 192.168.1.30
17:12:00.600 -> [EWS] headerName: Connection
17:12:00.600 -> [EWS] headerValue: keep-alive
17:12:00.600 -> [EWS] headerName: Cache-Control
17:12:00.634 -> [EWS] headerValue: max-age=0
17:12:00.634 -> [EWS] headerName: Upgrade-Insecure-Requests
17:12:00.634 -> [EWS] headerValue: 1
17:12:00.668 -> [EWS] headerName: User-Agent
17:12:00.668 -> [EWS] headerValue: Mozilla/5.0 (Windows NT 6.1; Win
17:12:00.737 -> [EWS] headerName: Accept
17:12:00.737 -> [EWS] headerValue: text/html,ap
17:12:00.737 -> [EWS] headerName: Accept-Encoding
17:12:00.737 -> [EWS] headerValue: gzi
17:12:00.737 -> [EWS] headerName: Accept-Language
17:12:00.737 -> [EWS] headerValue: en-
17:12:00.737 -> [EWS] headerName: Cookie
17:12:00.771 -> [EWS] headerValue: ESPSESSIONID
17:12:00.771 -> [EWS] args:  
17:12:00.771 -> [EWS] args count:  0
17:12:00.771 -> [EWS] args:  
17:12:00.771 -> [EWS] args count:  0
17:12:00.771 -> [EWS] Request: /
```

aagiannidis, 2021-11-06:
Hi all,

using the simple authentication example with the addition of the MemoryUsage library to show memory stats, one can see that over successive page reloads the memory heap is being exhausted...bringing everything to a stall..

```
19:23:41.443 -> Stack start:8670
19:23:41.443 -> Stack end:8703
19:23:41.443 -> Stack size:33
19:23:41.443 -> Data start:512
19:23:41.443 -> Heap start:3055
19:23:41.443 -> Heap end:4003

19:24:35.899 -> Stack start:8670
19:24:35.899 -> Stack end:8703
19:24:35.899 -> Stack size:33
19:24:35.899 -> Data start:512
19:24:35.899 -> Heap start:3055
19:24:35.899 -> Heap end:5186

19:25:43.502 -> Stack start:8670
19:25:43.502 -> Stack end:8703
19:25:43.502 -> Stack size:33
19:25:43.502 -> Data start:512
19:25:43.502 -> Heap start:3055
19:25:43.502 -> Heap end:7177

19:26:32.209 -> Stack start:8670
19:26:32.209 -> Stack end:8703
19:26:32.209 -> Stack size:33
19:26:32.209 -> Data start:512
19:26:32.209 -> Heap start:3055
19:26:32.209 -> Heap end:8122

19:27:05.778 -> Stack start:8670
19:27:05.778 -> Stack end:8703
19:27:05.778 -> Stack size:33
19:27:05.778 -> Data start:512
19:27:05.778 -> Heap start:3055
19:27:05.778 -> Heap end:8417
```

khoih-prog, 2021-11-06:
Hi @aagiannidis 

Thanks for your interests and nice words about the library.

I'm **amazed** that you've spent lots of time to investigate the issue. But my feeling is that you're using the worst board for the job (**Mega with only 16MHz, 8KB of RAM**) which can create problems (**heap corruption**, not enough memory, etc.) in real-time and it's very time-consuming and not worthy to spend time to investigate and solve the issue

As you know, this library also relies on so many underlying libraries / core (Ethernet library, AVR core, etc) , so any bug can also cause the issue.

I suggest you use a better board for the demanding job, such as **ESP32 / ESP8266 / Nano_RP2040_Connect** for WiFi, or **WT32_ETH01** for Ethernet (using [**WebServer_WT32_ETH01** library](https://github.com/khoih-prog/WebServer_WT32_ETH01)), and finally find out it's much more economical to use them as our time is much more precious.

Also have a look at [**Arduino hangs after repeated refreshing** #8](https://github.com/khoih-prog/EthernetWebServer/issues/8) and understand that using wrong board, with **less memory, slow**, etc., can create big headache

I'm sorry I have to close the issue, and won't spend time on this, even this is the real issue on the antique Mega board.

Good Luck,

closed 2021-11-06 by khoih-prog