khoih-prog/EthernetWebServer #43: server.authenticate  problem

dirkdickler, 2021-12-27:
If I use newest ver. 1.8.1  (when I use 1.7.0 work correct) 
- I get problem with  server.authenticate
- server.on(F("/"), []()  {
    if (!server.authenticate(www_username, www_password))
    {
      return server.requestAuthentication();
    }
    server.send(200, F("text/plain"), F("Login OK"));
  });

If you call above I don't get  LOGIN windows, but clear window 

- I testing on ESP32-S2 + W5500 + Ehtrenet3 lib  (this lib I patchet library)  
-

khoih-prog, 2021-12-27:
Close via [PR #42](https://github.com/khoih-prog/EthernetWebServer/pull/42)

closed 2021-12-27 by khoih-prog

khoih-prog, 2021-12-27:
Hi @dirkdickler  

The new [**EthernetWebServer releases v1.8.2**](https://github.com/khoih-prog/EthernetWebServer/releases/tag/v1.8.2) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/EthernetWebServer#contributions-and-thanks)

Best Regards,

---

### Releases v1.8.2

1. Fix wrong http status header bug. Check [fix for wrong http status header #42](https://github.com/khoih-prog/EthernetWebServer/pull/42)