khoih-prog/EthernetWebServer #44: multiple definition of base64 error with ver. 1.8.3

Nourbakhsh-Rad, 2022-01-11:
Hi,
After upgrading to version 1.8.3, the following error is observed during compilation

```
multiple definition of base64_init_encodestate
multiple definition of base64_encode_value
multiple definition of base64_encode_block
multiple definition of base64_encode_blockend
multiple definition of base64_encode_chars

```
I am using ESP-12E with ENC28J60 Ethernet module and Arduino IDE version 1.8.16 with ESP8266 core 3.0.2

khoih-prog, 2022-01-11:
Hi @Nourbakhsh-Rad 

Thanks for the bug report. I just forgot that ESP8266 already has the `lib64` in the core and then the

```
#if !(ESP32 || ESP8266)

...

#endif
```

in the files [**cdecode.c**](https://github.com/khoih-prog/EthernetWebServer/blob/master/src/libb64/cdecode.c) and [**cencode.c**](https://github.com/khoih-prog/EthernetWebServer/blob/master/src/libb64/cencode.c).

You can add those lines in the above-mentioned files if can't wait.

Will update the master and publish a new release soon.

Regards,

khoih-prog, 2022-01-11:
Hi @Nourbakhsh-Rad 

The new [**EthernetWebServer releases v1.8.4**](https://github.com/khoih-prog/EthernetWebServer/releases/tag/v1.8.4) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/EthernetWebServer#contributions-and-thanks)

Best Regards,

---

### Releases v1.8.4

1. Fix libb64 compile error for ESP8266. Check [**multiple definition of base64 error with ver. 1.8.3** #44](https://github.com/khoih-prog/EthernetWebServer/issues/44)
2. Update `Packages' Patches`

closed 2022-01-11 by khoih-prog

Nourbakhsh-Rad, 2022-01-11:
Thank you for your prompt action ..... @khoih-prog