khoih-prog/EthernetWebServer #53: Udp Multicast

kbssa, 2022-03-10:
Hello,

I am trying to make the UdpSendReceive example work in Multicast mode, it receive all the packets I send, but it never sends the reply message.

The unicast mode works as it should.

I just changed the line Udp.begin(localPort); to Udp.beginMulticast(IPAddress(23,255,255,250),1900);

```
#include "defines.h"

unsigned int localPort = 1900;    //10002;  // local port to listen on

char packetBuffer[255];          // buffer to hold incoming packet
char ReplyBuffer[] = "ACK";      // a string to send back

// A UDP instance to let us send and receive packets over UDP
EthernetUDP Udp;

void setup()
{
    Serial.begin(115200);
    while (!Serial);

    Serial.print("\nStart UDPSendReceive on "); Serial.print(BOARD_NAME);
    Serial.print(F(" with ")); Serial.println(SHIELD_TYPE);
    Serial.println(ETHERNET_WEBSERVER_VERSION);

    Ethernet.init(5);

    // start the ethernet connection and the server:
    // Use DHCP dynamic IP and random mac
    uint16_t index = millis() % NUMBER_OF_MAC;
    // Use Static IP
    //Ethernet.begin(mac[index], ip);
    Ethernet.begin(mac[index]);

    if (Ethernet.hardwareStatus() == EthernetNoHardware)
    {
        Serial.println("No Ethernet found. Stay here forever");

        while (true)
        {
            delay(1); // do nothing, no point running without Ethernet hardware
        }
    }

    if (Ethernet.linkStatus() == LinkOFF)
    {
        Serial.println("Not connected Ethernet cable");
    }

    Serial.print(F("Using mac index = "));
    Serial.println(index);

    Serial.print(F("Connected! IP address: "));
    Serial.println(Ethernet.localIP());

    Serial.println(F("\nStarting connection to server..."));
    // if you get a connection, report back via serial:
    Udp.beginMulticast(IPAddress(23,255,255,250),1900);

    Serial.print(F("Listening on port "));
    Serial.println(localPort);
}

void loop()
{
    // if there's data available, read a packet
    int packetSize = Udp.parsePacket();

    if (packetSize)
    {
        Serial.print(F("Received packet of size "));
        Serial.println(packetSize);
        Serial.print(F("From "));
        IPAddress remoteIp = Udp.remoteIP();
        Serial.print(remoteIp);
        Serial.print(F(", port "));
        Serial.println(Udp.remotePort());

        // read the packet into packetBufffer
        int len = Udp.read(packetBuffer, 255);

        if (len > 0)
        {
            packetBuffer[len] = 0;
        }

        Serial.println(F("Contents:"));
        Serial.println(packetBuffer);

        // send a reply, to the IP address and port that sent us the packet we received
        Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
        Udp.write(ReplyBuffer, strlen(ReplyBuffer));
        Udp.endPacket();
    }
}
```

Is there something missing in the code to make the multicast works ?

khoih-prog, 2022-03-10:
First, you have to understand Multicast and Multicast addresses by reading

1. [**Multicast_address**](https://en.wikipedia.org/wiki/Multicast_address)
2. [FAQ for Multicasting](https://web.archive.org/web/20110516204230/http://www.multicasttech.com/faq/)
3. [ClassD Multicast Address](https://web.archive.org/web/20110516204230/http://www.multicasttech.com/faq/#CLASSD)

> **Local [subnetwork](https://en.wikipedia.org/wiki/Subnetwork)**
> Addresses in the range of 224.0.0.0 to 224.0.0.255 are individually assigned by IANA and designated for multicasting on the local subnetwork only. For example, the [Routing Information Protocol](https://en.wikipedia.org/wiki/Routing_Information_Protocol) (RIPv2) uses 224.0.0.9, [Open Shortest Path First](https://en.wikipedia.org/wiki/Open_Shortest_Path_First) (OSPF) uses 224.0.0.5 and 224.0.0.6, and [Multicast DNS](https://en.wikipedia.org/wiki/Multicast_DNS) uses 224.0.0.251. Routers must not forward these messages outside the subnet from which they originate.

Some more examples how to use `UDP Multicast` can be found in [**MDNS_Generic** library](https://github.com/khoih-prog/MDNS_Generic/blob/master/src/MDNS_Generic_Impl.h)

```
#define  MDNS_SERVER_PORT        (5353)
...
static IPAddress mdnsMulticastIPAddr = IPAddress(224, 0, 0, 251);
...
statusCode = this->_udp->beginMulticast(mdnsMulticastIPAddr, MDNS_SERVER_PORT);
```

---

Second, you have to at least follow the instructions when creating an issue, with **critical information** such as

- Board name and type
- Ethernet board
- Ethernet library

because **some Ethernet boards and libraries are not supporting Multicast at all.**

Check [**UDP Multicast supported Ethernet modules / libraries**](https://github.com/khoih-prog/MDNS_Generic#currently-supported-ethernet-shieldsmodules)

Without the necessary information, the vague issue will be ignored and deleted next time.

Good Luck,

closed 2022-03-10 by khoih-prog

khoih-prog, 2022-03-10:
If you'd like to use UPnP / SSDP, try my [UPnP_Generic library](https://github.com/khoih-prog/UPnP_Generic) to be sure your system (router, network, etc.) is working OK.

https://github.com/khoih-prog/UPnP_Generic/blob/master/src/UPnP_Generic_Impl.h#L805-L852

```
#define UPNP_SSDP_PORT                    1900
...
IPAddress ipMulti(239, 255, 255, 250);            // multicast address for SSDP
...

bool UPnP::connectUDP()
{

#if UPNP_USING_ETHERNET
  #if USE_BUILTIN_ETHERNET
    #warning Using LAN8742A Ethernet and STM32Ethernet Lib in UPnP
    // For STM32 built-in LAN8742A Ethernet using STM32Ethernet Lib
    // initialize, start listening on specified port. IPAddress here must be local, not ipMulti address
    // virtual uint8_t beginMulticast(IPAddress, uint16_t);
    if (_udpClient.beginMulticast(Ethernet.localIP(), UPNP_SSDP_PORT))
    {
      return true;
    }
  #else
    // initialize, start listening on specified port.
    // virtual uint8_t beginMulticast(IPAddress, uint16_t);
    if (_udpClient.beginMulticast(ipMulti, UPNP_SSDP_PORT))
    {
      return true;
    }
  #endif
  
#elif UPNP_USING_WT32_ETH01
  if (_udpClient.beginMulticast(ipMulti, UPNP_SSDP_PORT))
  {
    return true;
  }
#else

  #if defined(ESP8266)
    if (_udpClient.beginMulticast(WiFi.localIP(), ipMulti, UPNP_SSDP_PORT))
    {
      return true;
    }
  #else
    if (_udpClient.beginMulticast(ipMulti, UPNP_SSDP_PORT))
    {
      return true;
    }
  #endif

#endif    //UPNP_USING_ETHERNET

  //UPNP_LOGINFO(F("UDP connection failed"));
  UPNP_LOGINFO0(F("F"));

  return false;
}

```