khoih-prog/EthernetWebServer #54: AdvancedWebServer example give me wrong ip HTTP EthernetWebServer is @ IP : 1.0.1.0

karadevnet, 2022-03-10:
i download and try to use a lib from you and simple example with RPI2040 PICO evb board with w5100. i read the code taht you write for spi pins for this evb board. serial monitor say that is connected but ip is not of dhcp or static that i try to give to make web server active in my network
here is some info from serial monitor :

Starting AdvancedWebServer on MBED RASPBERRY_PI_PICO with W5x00 using EthernetLarge Library
EthernetWebServer v2.0.1
[EWS] =========== USE_ETHERNET_LARGE ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 5
=========================
Currently Used SPI pinout:
MOSI:19
MISO:16
SCK:18
SS:17
=========================
Using mac index = 17
Connected! IP address: 0.1.0.1
HTTP EthernetWebServer is @ IP : 1.0.1.0
.......... .......... .......... ..

waiting for response. my router is mikrotik and network is simple like 192.168.88.1-254 but no success

khoih-prog, 2022-03-10:
I'm testing here and everything is OK

Be sure to connect the wires correctly as follows

1. For MBED mbed_rp2040 core

**MOSI:19
MISO:16
SCK:18
SS: 5**

```
Starting AdvancedWebServer on MBED RASPBERRY_PI_PICO with W5x00 using EthernetLarge Library
EthernetWebServer v2.0.1
[EWS] =========== USE_ETHERNET_LARGE ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 5
_pinCS = 0
W5100 init, using SS_PIN_DEFAULT = 17, new ss_pin = 10, W5100Class::ss_pin = 5
W5100::init: W5500, SSIZE =8192
=========================
Currently Used SPI pinout:
MOSI:19
MISO:16
SCK:18
SS:17
=========================
Using mac index = 17
Connected! IP address: 192.168.2.83
HTTP EthernetWebServer is @ IP : 192.168.2.83
.......... ....
```

2. For arduino-pico core

**MOSI:19
MISO:16
SCK:18
SS: 17**

```
Starting AdvancedWebServer on RASPBERRY_PI_PICO with W5x00 using EthernetLarge Library
EthernetWebServer v2.0.0
[EWS] =========== USE_ETHERNET_LARGE ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 17
_pinCS = 0
W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 17
W5100::init: W5500, SSIZE =8192
=========================
Currently Used SPI pinout:
MOSI:19
MISO:16
SCK:18
SS:17
=========================
Using mac index = 11
Connected! IP address: 192.168.2.82
HTTP EthernetWebServer is @ IP : 192.168.2.82
```
---

If you still have issue, recheck the wiring and/or replace the router / Ethernet board.

Good Luck,

closed 2022-03-10 by khoih-prog

khoih-prog, 2022-03-10:
If you'd like to use the CS/SS pin GPIO17, so that you don't need to change the wiring when switching between [**mbed_rp2040**](https://github.com/arduino/ArduinoCore-mbed) and [**arduino-pico**](https://github.com/earlephilhower/arduino-pico) cores, just modify

https://github.com/khoih-prog/EthernetWebServer/blob/5247b0ccd3f6ac910c131a6385e42d2f0ce95225/examples/AdvancedWebServer/defines.h#L276

to

```
 #define USE_THIS_SS_PIN       17
```

So you can use

**MOSI:19
MISO:16
SCK:18
SS: 17**

for both cores.