khoih-prog/EthernetWebServer #55: Teensy 4.1 server.streamFile not working

piranha771, 2022-04-19:
* PlatformIO Core [5.2.5] · Home [3.4.1]
* - Platform: Teensy 4.15.0
* Builder: Windows 10 21H1
* EWS 2.1.2

### Describe the bug

I try to send a html (actually a whole SPA) from an SD card over ethernet. But streaming files is not working.

### Steps to Reproduce

* Get any SD card.
* Put a file on it
* Get your teensy 4.1 and install EthernetWebServer.
* Apply patches, insert SD Card
* Init the built-in SD card library (SD.begin(BUILTIN_SDCARD));
* At any point in handler function insert:
```
if (SD.exists(path_to_file.c_str()))
{
    File myFile = SD.open(path_to_file.c_str());
    uint32_t myFileSize = myFile .size();
    Serial.println("SIZE to send: " + String(myFileSize)); // Debug check size
    server.streamFile(myFile , getContentType(path_to_file));
    myFile .close();
    return;
 }```

### Expected behavior

The `myFile` is send to the client

### Actual behavior

The debug check size output shows correct size of file to send.
The browser fails with `net::ERR_CONTENT_LENGTH_MISMATCH 200 (OK)`
(File is incomplete)

### Debug and AT-command log (if applicable)

```
Connected! IP address: 192.168.1.109
[EWS] handleClient: New Client
[EWS] method:  GET
[EWS] url:  /js/61.28ba07b9.js
[EWS] search:
[EWS] headerName: Host
[EWS] headerValue: 192.168.1.109
[EWS] headerName: Connection
[EWS] headerValue: keep-alive
[EWS] headerName: Upgrade-Insecure-Requests
[EWS] headerValue: 1
[EWS] headerName: User-Agent
[EWS] headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/100.0.4896.88 Safari/537.36
[EWS] headerName: Accept
[EWS] headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
[EWS] headerName: Accept-Encoding
[EWS] headerValue: gzip, deflate
[EWS] headerName: Accept-Language
[EWS] headerValue: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
[EWS] args:
[EWS] args count:  0
[EWS] args:
[EWS] args count:  0
[EWS] Request: /js/61.28ba07b9.js
[EWS] Arguments:
[EWS] Final list of key/value pairs:
[EWS] _handleRequest: request handler not found
SIZE to send: 1834

[EWS] send1: len =  0
[EWS] content =
[EWS] _prepareHeader response = HTTP/1.1 200 OK

[EWS] _prepareHeader sendHeader Conn close
[EWS] handleClient: Don't keepCurrentClient
```

khoih-prog, 2022-04-19:
FS is not supported for any board, but ESP32 and ESP8266 using native SPIFFS or LIttleFS.

This won't be worked on and supported in the near future.

closed 2022-04-19 by khoih-prog

piranha771, 2022-04-20:
I would it find more intuitive if the readme.md would contain a hint that this library only supports sending http content in form of strings and not streams (except ESP controllers). Thank you for your work nonetheless.