khoih-prog/EthernetWebServer #59: RP2040-W5500 UdpSendReceive

CanGuveren, 2022-07-22:
I am using RP20408(Arduino-Mbed core) and W5500. I tried UdpSendReceive but I can't work it. 

Start UDPSendReceive on MBED RASPBERRY_PI_PICO with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer v2.2.0
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 5
[ETG] W5100::init: no chip :-(
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 3
[EWS] MISO: 4
[EWS] SCK: 2
[EWS] SS: 5
[EWS] =========================
Using mac index = 11
Connected! IP address: 0.0.0.0

khoih-prog, 2022-07-22:
Hi @CanGuveren 

Please update to the latest [**mbed_rp2040 core v3.1.1**](https://github.com/arduino/ArduinoCore-mbed/releases/tag/3.1.1)

I've just tested and OK here

```
Start UDPSendReceive on MBED RASPBERRY_PI_PICO with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer v2.2.0
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 17
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
Using mac index = 7
Connected! IP address: 192.168.2.113

Starting connection to server...
Listening on port 1883
```

closed 2022-07-22 by khoih-prog

CanGuveren, 2022-07-25:
I use MOSI: 3, MISO: 4, SCK: 2, SS: 5. How can I change SPI pins?

Sireevenkat, 2022-07-27:
I am also facing same issue as @CanGuveren mentioned. I am  getting IP address :0.0.0.0.I am using ESP32 and Arduino UNO boards with W5500 Ethernet shield. How can I resolve this issue