khoih-prog/EthernetWebServer #62: AdvancedWebServer example give me wrong ip HTTP EthernetWebServer is @ IP : 0.0.0.0

proasnet, 2022-08-24:
### Describe the bug

I have a new board SAMD21 M0 size as Arduino UNO, compatible with Arduino M0, native port, with ethernet shield Wiznet W5100.
I tried example code Hello World, edit all serial.print to serialUSB.print and load to cpu over Native USB port, I have selected Arduino M0 board in Arduino IDE. Then, I get this from serial monitor:
```
Starting HelloServer on SAMD Zero with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer v2.2.0
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
[EWS] Board : SAMD Zero , setCsPin: 10
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
Using mac index = 12
Connected! IP address: 0.0.0.0
```
I have a company network on 192.168.0.x with DHCP. Before, this shield I used with MEGA and webduino, working.
Thank you.

khoih-prog, 2022-08-24:
> Connected! IP address: 0.0.0.0

This means the W5100 board is not connected correctly and is not initialized.

> [EWS] Board : SAMD Zero , setCsPin: 10

Show how you connected (M0 pin <-> W5100)

khoih-prog, 2022-08-24:
Try the simplest [**WebClient** example](https://github.com/arduino-libraries/Ethernet/tree/master/examples/WebClient) from [**Ethernet** library](https://github.com/arduino-libraries/Ethernet) and **be sure working** before moving to more complex libraries and examples.

I'm closing the issue now, and won't reopen until you prove this is a bug of this library.

closed 2022-08-24 by khoih-prog

proasnet, 2022-08-25:
Thank you for reply. I tried this with SAMD21 M0 board and with Arduino UNO. Both with the same ethernet shield and WebClient example from Ethernet Library.
This is monitor from UNO:

```
Starting WebClient on AVR Mega with W5x00 using Ethernet_Generic Library with Large Buffer
Ethernet_Generic v2.4.0
[ETG] Default SPI pinout:
[ETG] MOSI: 11
[ETG] MISO: 12
[ETG] SCK: 13
[ETG] SS: 10
[ETG] =========================
[ETG] Board : AVR Mega , setCsPin: 10
[ETG] W5100 init, using SS_PIN_DEFAULT = 10 , new ss_pin =  10 , W5100Class::ss_pin =  10
[ETG] Chip is W5100
[ETG] W5100::init: W5100, SSIZE = 4096
[ETG] Currently Used SPI pinout:
[ETG] MOSI: 11
[ETG] MISO: 12
[ETG] SCK: 13
[ETG] SS: 10
[ETG] =========================
Using mac index = 18
Connected! IP address: 192.168.0.172

Starting connection to server...
Connected to server
HTTP/1.1 200 OK
Date: Thu, 25 Aug 2022 05:13:59 GMT
Content-Type: text/plain
Content-Length: 2263
Connection: close
x-amz-id-2: AmeituDrQX407ke3Rb+od+W+hfIgUGdK3yhMo0VTer1Vn9pD6YHGZzaoySPwtP+1YzE9igw6+N4=
x-amz-request-id: RKWZFAZEM0Y9W9VG
Last-Modified: Wed, 23 Feb 2022 14:56:42 GMT
ETag: "667cf48afcc12c38c8c1637947a04224"
CF-Cache-Status: DYNAMIC
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=MjBJ9H50ZNvNB98g99a4xBHFHMiaqEdbaSX%2BpEe7yvk%2BjoGz46%2FQiYDZSnxZpQWIsR5e8%2F725XF9gXJrQ12M98917cDT5EbBWPymqdXsUlaCmaVCPDTQCHmW65mquZg%3D"}],"group":"cf-nel","max_age":604800}
NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Server: cloudflare
CF-RAY: 7401d772ad2db33c-PRG
alt-svc: h3=":443"; ma=86400, h3-29=":443"; ma=86400

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    
   ;;;;;;;;:`   `;;;;;;;;;     ,;;;;;;;;.`   .;;;;;;;;   
  .;;;;;;,         :;;;;;;;   .;;;;;;;          ;;;;;;;  
  ;;;;;;             ;;;;;;;  ;;;;;;,            ;;;;;;. 
 ,;;;;;               ;;;;;;.;;;;;;`              ;;;;;; 
 ;;;;;.                ;;;;;;;;;;;`      ```       ;;;;;`
 ;;;;;                  ;;;;;;;;;,       ;;;       .;;;;;
`;;;;:                  `;;;;;;;;        ;;;        ;;;;;
,;;;;`    `,,,,,,,,      ;;;;;;;      .,,;;;,,,     ;;;;;
:;;;;`    .;;;;;;;;       ;;;;;,      :;;;;;;;;     ;;;;;
:;;;;`    .;;;;;;;;      `;;;;;;      :;;;;;;;;     ;;;;;
.;;;;.                   ;;;;;;;.        ;;;        ;;;;;
 ;;;;;                  ;;;;;;;;;        ;;;        ;;;;;
 ;;;;;                 .;;;;;;;;;;       ;;;       ;;;;;,
 ;;;;;;               `;;;;;;;;;;;;                ;;;;; 
 `;;;;;,             .;;;;;; ;;;;;;;              ;;;;;; 
  ;;;;;;:           :;;;;;;.  ;;;;;;;            ;;;;;;  
   ;;;;;;;`       .;;;;;;;,    ;;;;;;;;        ;;;;;;;:  
    ;;;;;;;;;:,:;;;;;;;;;:      ;;;;;;;;;;:,;;;;;;;;;;   
    `;;;;;;;;;;;;;;;;;;;.        ;;;;;;;;;;;;;;;;;;;;    
      ;;;;;;;;;;;;;;;;;           :;;;;;;;;;;;;;;;;:     
       ,;;;;;;;;;;;;;,              ;;;;;;;;;;;;;;       
         .;;;;;;;;;`                  ,;;;;;;;;:         
                                                         
                                                         
                                                         
                                                         
    ;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   

Disconnecting from server...
```
This is monitor from SAMD21 M0:
```
Starting WebClient on SAMD Zero with W5x00 using Ethernet_Generic Library with Large Buffer
Ethernet_Generic v2.4.0
[ETG] Default SPI pinout:
[ETG] MOSI: 23
[ETG] MISO: 22
[ETG] SCK: 24
[ETG] SS: 16
[ETG] =========================
[ETG] Board : SAMD Zero , setCsPin: 10
[ETG] W5100 init, using SS_PIN_DEFAULT = 10 , new ss_pin =  10 , W5100Class::ss_pin =  10
[ETG] W5100::init: no chip :-(
[ETG] Currently Used SPI pinout:
[ETG] MOSI: 23
[ETG] MISO: 22
[ETG] SCK: 24
[ETG] SS: 16
[ETG] =========================
Using mac index = 3
Connected! IP address: 0.0.0.0
Speed: NO LINK, Duplex: NO LINK, Link status: NO LINK

Starting connection to server...

Disconnecting from server...
```

khoih-prog, 2022-08-25:
You have to modify these lines 73-80  from

https://github.com/khoih-prog/EthernetWebServer/blob/4fa2bca31c9442cb31a1b4af42cefc2f5c98e6d8/examples/HelloServer/defines.h#L73-L80

to

```
#if defined(ETHERNET_USE_SAMD)
  // For SAMD
  // Default pin 10 to SS/CS
  #define USE_THIS_SS_PIN       10
  
  #if ( defined(ARDUINO_SAMD_ZERO) && !defined(SEEED_XIAO_M0) )
    #undef USE_THIS_SS_PIN
    #define USE_THIS_SS_PIN       SS
    #define BOARD_TYPE      "SAMD Zero"
  #elif defined(ARDUINO_SAMD_MKR1000)
```

If not working, you have to connect directly from `ICSP` connector of W5100 to M0 pins according

[image]

|W5100 ICSP|<--->|M0|
|:-:|:-:|:-:|
|MOSI|<--->|23|
|MISO|<--->|22|
|SCK|<--->|24|
|SS|<--->|16|

If not working, you can ask on Arduino Forum for help (M0 card bad SPI pins, M0 bad ICSP connection, etc)

Good Luck,

khoih-prog, 2022-08-26:
You can also try the new [**EthernetWebServer** v2.2.1](https://github.com/khoih-prog/EthernetWebServer/releases/tag/v2.2.1), which will try to auto-select the `SS/CS` pin according to `board package`

---

### Releases v2.2.1

1. Auto-select SPI(s) `SS/CS` pins according to board package if available
2. Update `Packages' Patches`

proasnet, 2022-08-26:
Hi, thank you for help.
I tried the first and second edit of defines.h, with the same result.
Monitor:
```
Starting HelloServer on SAMD Zero with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer v2.2.0
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
[EWS] Board : SAMD Zero , setCsPin: 16
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
Using mac index = 3
Connected! IP address: 255.255.255.255
```

khoih-prog, 2022-08-26:
Hi @proasnet 

It seems that your `SAMD21 M0` board's ICSP is not compatible with your W5100 **SPI** bus. I don't have that kind of weird board to test and help.

If you have the multimeter, you can check the M0 ICSP is connecting to which pins.

The remaining way is to read the docs, ask around the Arduino Forum, to match the SPI pins `23, 23, 24 and 16` to your `SAMD21 M0` to `D0-D13, AREF, SDA, SCL, A0-A5`

Somewhere you can find the SPI pins used for that special M0 board.

khoih-prog, 2022-08-26:
You can compare with the schematic at

https://robotdyn.com/pub/media/0G-00005517==SAMD21-M0/DOCS/Schematic==0G-00005517==SAMD21-M0.pdf

[image]

https://robotdyn.com/pub/media/0G-00005517==SAMD21-M0/DOCS/PINOUT==0G-00005517==SAMD21-M0.pdf
[image]

proasnet, 2022-08-26:
Its the same
[arduino-M0-schematic.pdf](https://github.com/khoih-prog/EthernetWebServer/files/9430121/arduino-M0-schematic.pdf)
I only found that this board HW-819 is compatible with Arduino M0. I not found schematic from producer of my board HW-819.
I wonder if the SAMD21 G18 CPU needs to define SPI with the help of SERCOM, but I have no experience with it yet and I can't do it. Yesterday I solved SERCOM for UART on my board, because I need 4xUART for my project and so far I'm just trying to set it up. I'm not an advanced programmer, you can say that I'm still a beginner.
.

khoih-prog, 2022-08-26:
For M0,

Pin  10   <=>  PA18 

Pin  16   <=>  PB09  <=>  A2
Pin  22   <=>  PA12  <=>  MISO
Pin  23   <=>  PB10  <=>  MOSI
Pin  24   <=>  PB11  <=>  SCK

With your latest test, it seems that **M0 is communicating OK with W5100 (no more no chip error).**

```
[EWS] =========================
[EWS] Board : SAMD Zero , setCsPin: 16
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
Using mac index = 3
Connected! IP address: 255.255.255.255

Starting connection to server...
```

Possibly M0 is bad (SPI, etc.) It's better you buy W5500 and connect directly to M0 to try.

I suggest you use RP2040 + W5100S or W5500, such as WIZNET_5500_EVB_PICO, WIZNET_5100S_EVB_PICO. More powerful and integrated.

proasnet, 2022-08-26:
I have another shield for test SPI, USB HOST 2.0, I try it.

proasnet, 2022-08-26:
So, my next experimet was, that I use a example Ethernet 2.0.1 library, WebClient, this working.
Monitor:
```
connecting to www.google.com...
connected to 142.251.36.100
HTTP/1.1 200 OK
Content-Type: text/html; charset=ISO-8859-1
Date: Fri, 26 Aug 2022 06:21:36 GMT
Expires: -1
Cache-Control: private, max-age=0
Content-Security-Policy: object-src 'none';base-uri 'self';script-src 'nonce-pn68xgpXn2nyOKTmfZqC7Q' 'strict-dynamic' 'report-sample' 'unsafe-eval' 'unsafe-inline' https: http:;report-uri https://csp.withgoogle.com/csp/gws/xsrp
Server: gws
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN
Set-Cookie: AEC=AakniGN7dU2ewEeqgp5ZHcufbM-y0RBGo-ZMXI99ycuCRfsDFTK0479yJzM; expires=Wed, 22-Feb-2023 06:21:36 GMT; path=/; domain=.google.com; Secure; HttpOnly; SameSite=lax
Accept-Ranges: none
Vary: Accept-Encoding
Connection: close
Transfer-Encoding: chunked
```

khoih-prog, 2022-08-26:
Good news, I'll check why we have M0 problem getting correct IP address.
Possibly recent update in [Ethernet_Generic v2.4.0](https://github.com/khoih-prog/Ethernet_Generic/releases/tag/v2.4.0) is not compatible with Arduino SAAMD core ???

> ### Releases v2.4.0
> 
> 1. Using **raw_address()** as default instead of **private IPAddress data**
> 2. Add missing `LibraryPatches` for ESP32
> 3. Update `Packages' Patches`

If you're willing to spend some time to help test and fix the bug (as I currently don't have the same SAMD21 M0 board), I'll reopen this issue.

-------------------

Steps to test

1. Install [**Ethernet_Generic v2.3.1**](https://github.com/khoih-prog/Ethernet_Generic/releases/tag/v2.3.1) to replace [**Ethernet_Generic v2.4.0**](https://github.com/khoih-prog/Ethernet_Generic/releases/tag/v2.4.0)
2. Use the same example code of this latest [**EthernetWebServer** v2.2.1](https://github.com/khoih-prog/EthernetWebServer/releases/tag/v2.2.1)

proasnet, 2022-08-26:
Hi, I can continue at next week at wednesday, because to this time I am at
home without a board and time to do it.

proasnet, 2022-08-31:
Hi @khoih-prog ,
now I tested example of EthernetWebServer - WebClient with versions of an Ethernet Generic, v. 2.3.1 and 2.5.0 with the same problem:
```
Starting WebClient on SAMD Zero with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer v2.2.1
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
[EWS] Board : SAMD Zero , setCsPin: 16
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] SS: 16
[EWS] =========================
Using mac index = 12
Connected! IP address: 0.0.0.0

Starting connection to server...

Disconnecting from server...
```

khoih-prog, 2022-08-31:
Can you use the the following test code (using Ethernet or EthernetLarge library)

#### 1) File WebClient_SAMD21_M0.ino

```
/****************************************************************************************************************************
  WebClient_SAMD21_M0.ino - Simple Arduino web server sample for Ethernet shield
  
  Ethernet_Generic is a library for the W5x00 Ethernet shields trying to merge the good features of
  previous Ethernet libraries
  
  Built by Khoi Hoang https://github.com/khoih-prog/Ethernet_Generic
 *****************************************************************************************************************************/

#include "defines.h"

char server[] = "arduino.tips";

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 0, 188);
IPAddress myDns(192, 168, 0, 1);

// Initialize the Ethernet client object
EthernetClient client;

void setup()
{
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  delay(500);

  Serial.print("\nStarting WebClient_SAMD21_M0 on "); Serial.print(BOARD_NAME);
  Serial.print(F(" with ")); Serial.println(SHIELD_TYPE); 
  //Serial.println(ETHERNET_GENERIC_VERSION);

  Serial.println(F("Default SPI pinout:"));
  Serial.print(F("MOSI:")); Serial.println(MOSI);
  Serial.print(F("MISO:")); Serial.println(MISO);
  Serial.print(F("SCK:")); Serial.println(SCK);
  Serial.print(F("SS:")); Serial.println(SS);
  Serial.println(F("========================="));

  // Use default SS = 10
  #ifndef USE_THIS_SS_PIN
    #define USE_THIS_SS_PIN   10    // For other boards
  #endif

    Serial.print(F("Board:")); Serial.print(BOARD_NAME); Serial.print(F(", setCsPin:")); Serial.println(USE_THIS_SS_PIN);

  // For other boards, to change if necessary 
  Ethernet.init (USE_THIS_SS_PIN);

  // start the ethernet connection and the server:
  // Use DHCP dynamic IP and random mac
  uint16_t index = millis() % NUMBER_OF_MAC;
  // Use Static IP
  //Ethernet.begin(mac[index], ip);
  //Ethernet.begin(mac[index]);
  // try to configure using IP address instead of DHCP:
  Ethernet.begin(mac[index], ip, myDns);

  // Just info to know how to connect correctly
  // To change for other SPI
  Serial.println(F("Currently SPI pinout:"));
  Serial.print(F("MOSI:")); Serial.println(MOSI);
  Serial.print(F("MISO:")); Serial.println(MISO);
  Serial.print(F("SCK:")); Serial.println(SCK);
  Serial.print(F("USE_THIS_SS_PIN:")); Serial.println(USE_THIS_SS_PIN);
  Serial.println(F("========================="));
[... 123 lines cut ...]
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x12 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x13 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x14 },
};

#endif    //defines_h
```

---

and see if OK. 

If OK, then to recheck with Ethernet_Generic library, modify a line in the `defines.h`

```
#include "Ethernet.h"
```

to

```
#include "Ethernet_Generic.h"
```

I need some more debug info as I won't have the M0 board here until several weeks (just ordered).

proasnet, 2022-09-01:
Hi @khoih-prog
Compile error:
```
Z:\Jirutka\99_VYVOJ ARDUINO\WebClient_test\WebClient_test.ino: In function 'void setup()':
WebClient_test:28:74: error: 'BOARD_NAME' was not declared in this scope
   SerialUSB.print("\nStarting WebClient_SAMD21_M0 on "); SerialUSB.print(BOARD_NAME);
                                                                          ^~~~~~~~~~
Z:\Jirutka\99_VYVOJ ARDUINO\WebClient_test\WebClient_test.ino:28:74: note: suggested alternative: '__ASMNAME'
   SerialUSB.print("\nStarting WebClient_SAMD21_M0 on "); SerialUSB.print(BOARD_NAME);
                                                                          ^~~~~~~~~~
                                                                          __ASMNAME
Byly nalezené násobné knihovny "Ethernet.h"
 Použitý: C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet
 Nepoužitý: C:\Program Files (x86)\Arduino\libraries\Ethernet
exit status 1
'BOARD_NAME' was not declared in this scope
```

khoih-prog, 2022-09-01:
You forget to patch the core.

Anyway, add to defines.h

```
#define BOARD_NAME     "SAMD21 M0"
```

---

Great. You did it. Then change to use `Ethernet_Generic` library

khoih-prog, 2022-09-01:
So, the issue is the W5100 Ethernet shield hardwires SS to pin 10, not using normal SS in the core.

proasnet, 2022-09-01:
I updated previous comment, please see up. Thanks

khoih-prog, 2022-09-01:
Try comment out to see if better ???

```
// For other boards, to change if necessary 
  Ethernet.init (USE_THIS_SS_PIN);
```

proasnet, 2022-09-01:
It not help, the same result.
```
Starting WebClient_SAMD21_M0 on SAMD21 M0 with W5x00 using Ethernet Library with Large Buffer
Ethernet_Generic v2.5.0
Default SPI pinout:
MOSI:23
MISO:22
SCK:24
SS:16
=========================
Currently SPI pinout:
MOSI:23
MISO:22
SCK:24
USE_THIS_SS_PIN:10
=========================
Using mac index = 11
Connected! IP address: 0.0.0.0

Starting connection to server...

Disconnecting from server...
```

khoih-prog, 2022-09-01:
OK, I think I found out the issue.

Try to add to `defines.h` before 

```
#include "Ethernet_Generic.h"
```

such as

```
#define USE_W5100     true
...
#include "Ethernet_Generic.h"
```

proasnet, 2022-09-01:
in ino:
`//Ethernet.init (USE_THIS_SS_PIN);`
in defines:
```
#define USE_W5100     true
#include "Ethernet_Generic.h"
```
in monitor:
```
Starting WebClient_SAMD21_M0 on SAMD21 M0 with W5x00 using Ethernet Library with Large Buffer
Ethernet_Generic v2.5.0
Default SPI pinout:
MOSI:23
MISO:22
SCK:24
SS:16
=========================
Currently SPI pinout:
MOSI:23
MISO:22
SCK:24
USE_THIS_SS_PIN:10
=========================
Using mac index = 12
Connected! IP address: 0.0.0.0

Starting connection to server...

Disconnecting from server...
```

proasnet, 2022-09-01:
Dear @khoih-prog ,
only as note, if this working with ethernet.h, is any reason for unuse it?
Only for my project, I am not need universal ethernet library.
Of course I'll help you debug the library.

khoih-prog, 2022-09-01:
> only as note, if this working with ethernet.h, is any reason for unuse it?
> Only for my project, I am not need universal ethernet library.
> Of course I'll help you debug the library.

Thanks.

---

Sorry, one more step before we have to wait

Change

https://github.com/khoih-prog/Ethernet_Generic/blob/main/src/utility/w5100.h#L72

from

```
  #define SPI_ETHERNET_SETTINGS SPISettings(14000000, MSBFIRST, SPI_MODE0)
```

to

```
#define SPI_ETHERNET_SETTINGS SPISettings(8000000, MSBFIRST, SPI_MODE0)
```

proasnet, 2022-09-01:
This header file I found in my computer at
C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet_Generic\src\utility
and I changed it. Its the right way?
monitor:
```
Starting WebClient_SAMD21_M0 on SAMD21 M0 with W5x00 using Ethernet Library with Large Buffer
Ethernet_Generic v2.5.0
Default SPI pinout:
MOSI:23
MISO:22
SCK:24
SS:16
=========================
Board:SAMD21 M0, setCsPin:10
Currently SPI pinout:
MOSI:23
MISO:22
SCK:24
USE_THIS_SS_PIN:10
=========================
Using mac index = 11
Connected! IP address: 0.0.0.0

Starting connection to server...

Disconnecting from server...
```

khoih-prog, 2022-09-01:
The actual issue is that the W5100 is too old and it's better to use W5500.
The new Ethernet_Generic is the merge of several Ethernet libraries, such as Ethernet, EthernetLarge, Ethernet2 and Ethernet3 to deal with performance and support to many new Ethernet shleds, such as W5100S, Large buffer, faster clock, etc.

If you're using W5100, you can use Ethernet library, without problem.

proasnet, 2022-09-01:
Thank you for the information. I bought my shield 3 months ago, there are a lot of them in e-shops. I also looked at the W5500, but I couldn't find one for a normal price as a shield, which is not on the wires, but as a plug-in module.

proasnet, 2022-09-01:
I would also like to ask you if you would not modify the WebServer example with the ethernet.h library so that it is clean only for my SAMD21 M0 board, without the maze of else, else if endif, etc. It would take me a long time to clean it up .
Thank you very much

khoih-prog, 2022-09-01:
One more change, if you have time to help

Comment out

https://github.com/khoih-prog/Ethernet_Generic/blob/main/src/utility/w5100.h#L110

```
//#define SPI_ETHERNET_SETTINGS SPISettings(30000000, MSBFIRST, SPI_MODE3)
```

---

> I would also like to ask you if you would not modify the WebServer example with the ethernet.h library so that it is clean only for my SAMD21 M0 board, without the maze of else, else if endif, etc. It would take me a long time to clean it up .

Will do later

proasnet, 2022-09-01:
compile error:
```
In file included from C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet_Generic\src/utility/w5100_Impl.h:45:0,
                 from C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet_Generic\src/Ethernet_Generic_Impl.h:59,
                 from C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet_Generic\src/Ethernet_Generic.h:85,
                 from Z:\Jirutka\99_VYVOJ ARDUINO\WebClient_test\defines.h:38,
                 from Z:\Jirutka\99_VYVOJ ARDUINO\WebClient_test\WebClient_test.ino:10:
C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet_Generic\src/utility/w5100.h: In static member function 'static void W5100Class::beginSPITransaction()':
C:\Users\j.jirutka\Documents\Arduino\libraries\Ethernet_Generic\src/utility/w5100.h:546:34: error: 'SPI_ETHERNET_SETTINGS' was not declared in this scope
       pCUR_SPI->beginTransaction(SPI_ETHERNET_SETTINGS);
                                  ^~~~~~~~~~~~~~~~~~~~~
exit status 1
Nastala chyba při kompilaci u desky Arduino Zero (Native USB Port).
```

khoih-prog, 2022-09-01:
Sorry again, replace 

https://github.com/khoih-prog/Ethernet_Generic/blob/main/src/utility/w5100.h#L110

with

```
#define SPI_ETHERNET_SETTINGS SPISettings(8000000, MSBFIRST, SPI_MODE0)
```

proasnet, 2022-09-01:
Working:
```
Starting WebClient_SAMD21_M0 on SAMD21 M0 with W5x00 using Ethernet Library with Large Buffer
Ethernet_Generic v2.5.0
Default SPI pinout:
MOSI:23
MISO:22
SCK:24
SS:16
=========================
Board:SAMD21 M0, setCsPin:10
Currently SPI pinout:
MOSI:23
MISO:22
SCK:24
USE_THIS_SS_PIN:10
=========================
Using mac index = 1
Connected! IP address: 192.168.0.177

Starting connection to server...

Disconnecting from server...
```

khoih-prog, 2022-09-01:
Good news. That's really because of the too old W5100 requires **slow clock.**

I have another W5100 but never have this kind of slow-clock issue.

I'll make some notes and user configurable settings for W5100 in the new release.

proasnet, 2022-09-01:
Good. Thank you.
Please, when you have time, please make me an example for my board and W5100 as a webserver with basicAuth.

khoih-prog, 2022-09-01:
Will do.

khoih-prog, 2022-09-01:
HI @proasnet 

The new [**Ethernet_Generic releases v2.5.1**](https://github.com/khoih-prog/Ethernet_Generic/releases/tag/v2.5.1) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/Ethernet_Generic#contributions-and-thanks)

Best Regards,

---

### Releases v2.5.1

1. Slow SPI clock to `8MHz` and select `SPI_MODE0` for old W5100 shield using SAMD Zero. Check [**AdvancedWebServer example give me wrong ip HTTP EthernetWebServer is @ IP : 0.0.0.0** #62](https://github.com/khoih-prog/EthernetWebServer/issues/62)
2. Default to `W5100` to be safe. To change if using `W5200, W5500, W5100s` or testing faster SPI clock speeds

khoih-prog, 2022-09-01:
Hi @proasnet 

As promised, hereafter is the code for `HTTPBasicAuth_SAMD_M0` as requested. Just modify as needed accorfding to your use-case.

### HTTPBasicAuth_SAMD_M0

```
/****************************************************************************************************************************
  HTTPBasicAuth_SAMD_M0.h - Dead simple web-server for Ethernet shields

  EthernetWebServer is a library for the Ethernet shields to run WebServer

  Based on and modified from ESP8266 https://github.com/esp8266/Arduino/releases
  Built by Khoi Hoang https://github.com/khoih-prog/EthernetWebServer
  Licensed under MIT license
 *****************************************************************************************************************************/
/*
   The SAMD Zero can only use 8MHz clock and SPI_MODE0 for W5100
*/

#if !defined(ARDUINO_SAMD_ZERO)
  #error For SAMD21 M0 or Zero only
#endif

// For SAMD Zero
// Default pin SS/CS,if no SS pin, use pin 10
#define USE_THIS_SS_PIN       10    //SS

// Debug Level from 0 to 4
#define _ETHERNET_WEBSERVER_LOGLEVEL_       2

///////////////////////////////////////////////////////////

// W5100 chips can have up to 4 sockets.  W5200 & W5500 can have up to 8 sockets.
// Use EthernetLarge feature, Larger buffers, but reduced number of simultaneous connections/sockets (MAX_SOCK_NUM == 2)
#define ETHERNET_LARGE_BUFFERS

//////////////////////////////////////////////////////////

#include "EthernetWebServer.h"

#if defined(ETHERNET_LARGE_BUFFERS)
  #define SHIELD_TYPE           "W5x00 using Ethernet Library with Large Buffer"
#else
  #define SHIELD_TYPE           "W5x00 using Ethernet Library"
#endif

byte mac[] =  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 0, 188);
IPAddress myDns(8, 8, 8, 8);

EthernetWebServer server(80);

const char* www_username = "admin";
const char* www_password = "http_auth";

void setup()
{
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  Serial.print("\nStarting HTTPBasicAuth_SAMD_M0 on SAMD Zero with"); Serial.println(SHIELD_TYPE);
  Serial.println(ETHERNET_WEBSERVER_VERSION);

  ET_LOGWARN(F("SPI pinout:"));
  ET_LOGWARN1(F("MOSI:"), MOSI);
  ET_LOGWARN1(F("MISO:"), MISO);
  ET_LOGWARN1(F("SCK:"),  SCK);
  ET_LOGWARN1(F("CS:"),   USE_THIS_SS_PIN);
  ET_LOGWARN(F("========================="));

  Ethernet.init (USE_THIS_SS_PIN);
    
  // Use DHCP
  Ethernet.begin(mac);
  // Use Static IP
  //Ethernet.begin(mac, ip); 
  // Use static IP and DNS
  //Ethernet.begin(mac, ip, myDns);

  Serial.print(F("Connected! IP address: "));
  Serial.println(Ethernet.localIP());

  server.on(F("/"), []()
  {
    if (!server.authenticate(www_username, www_password))
    {
      return server.requestAuthentication();
    }
    server.send(200, F("text/plain"), F("Login OK"));
  });

  server.begin();

  Serial.print(F("Open http://")); Serial.print(Ethernet.localIP());
  Serial.println(F("/ in your browser to see it working"));
  Serial.print(F("Using username : ")); Serial.print(www_username);
  Serial.print(F(" and password : "));  Serial.println(www_password);
}

void loop()
{
  server.handleClient();
}
```

proasnet, 2022-09-08:
Hi @khoih-prog , I am sorry for later answer.
Now I testing code for me, this I get in monitor
```
Starting HTTPBasicAuth_SAMD_M0 on SAMD Zero withW5x00 using Ethernet Library with Large Buffer
EthernetWebServer v2.2.2
Connected! IP address: 0.0.0.0
Open http://0.0.0.0/ in your browser to see it working
Using username : admin and password : http_auth
```
Without IP adress again.
Thank you

khoih-prog, 2022-09-08:
Sorry, I think I forgot some lines of code because I didn't have the M0 board then. Please add at the beginning

```
#if !defined(ARDUINO_SAMD_ZERO)
  #error For SAMD21 M0 or Zero only
#endif
   
// Default to use W5100. Must change to false for W5500, W5100S, for faster SPI clock
// Must use true for SAMD21, such as Zero, SAMD_FEATHER_M0_EXPRESS, etc.
#define USE_W5100                           true

// Use this for ARDUINO_SAMD_ZERO, etc. if can't print to terminal with Serial.print
#if defined(SERIAL_PORT_USBVIRTUAL)
  #define Serial          SERIAL_PORT_USBVIRTUAL
  #warning Using SAMD Zero SerialUSB
#endif

// For SAMD Zero
// Default pin SS/CS,if no SS pin, use pin 10
#define USE_THIS_SS_PIN       10    //SS

// Debug Level from 0 to 4
#define _ETHERNET_WEBSERVER_LOGLEVEL_       2
...
```

PS: **Bad luck** that I just received the M0 board and it broken at the first try. So `great` quality. Have to order 2 more to keep for testing in the future.

[image]

proasnet, 2022-09-12:
Thank you @khoih-prog 
Today I tested it with inserted last code
I get this, without any IP adress
```
Starting HTTPBasicAuth_SAMD_M0 on SAMD Zero withW5x00 using Ethernet Library with Large Buffer
EthernetWebServer v2.2.2
[EWS] SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] CS: 10
[EWS] =========================
```

khoih-prog, 2022-09-13:
I'm sure missing some lines somewhere. Still OK with Ethernet_Generic's `WebClient_SAMD21_M0` ?

You have to wait some time until I receive the new boards ;={{

khoih-prog, 2022-09-26:
Hi @proasnet 

Just received the boards.

Try this

```
/****************************************************************************************************************************
  HTTPBasicAuth_SAMD_M0.h - Dead simple web-server for Ethernet shields

  EthernetWebServer is a library for the Ethernet shields to run WebServer

  Based on and modified from ESP8266 https://github.com/esp8266/Arduino/releases
  Built by Khoi Hoang https://github.com/khoih-prog/EthernetWebServer
  Licensed under MIT license
 *****************************************************************************************************************************/
/*
   The SAMD Zero can only use 8MHz clock and SPI_MODE0 for W5100
*/

#if !defined(ARDUINO_SAMD_ZERO)
  #error For SAMD21 M0 or Zero only
#endif

#define BOARD_TYPE      "SAMD Zero"
    
// Default to use W5100. Must change to false for W5500, W5100S, for faster SPI clock
// Must use true for SAMD21, such as Zero, SAMD_FEATHER_M0_EXPRESS, etc.
#define USE_W5100                           true

// Use this for ARDUINO_SAMD_ZERO, etc. if can't print to terminal with Serial.print
#if defined(SERIAL_PORT_USBVIRTUAL)
  #define Serial          SERIAL_PORT_USBVIRTUAL
  #warning Using SAMD Zero SerialUSB
#endif

// For SAMD Zero
// Default pin SS/CS,if no SS pin, use pin 10
#define USE_THIS_SS_PIN       10    //SS

// Debug Level from 0 to 4
#define _ETHERNET_WEBSERVER_LOGLEVEL_       2

///////////////////////////////////////////////////////////

// W5100 chips can have up to 4 sockets.  W5200 & W5500 can have up to 8 sockets.
// Use EthernetLarge feature, Larger buffers, but reduced number of simultaneous connections/sockets (MAX_SOCK_NUM == 2)
#define ETHERNET_LARGE_BUFFERS

//////////////////////////////////////////////////////////

#include "EthernetWebServer.h"

#if defined(ETHERNET_LARGE_BUFFERS)
  #define SHIELD_TYPE           "W5x00 using Ethernet Library with Large Buffer"
#else
  #define SHIELD_TYPE           "W5x00 using Ethernet Library"
#endif

byte mac[] =  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 0, 188);
IPAddress myDns(8, 8, 8, 8);

EthernetWebServer server(80);

const char* www_username = "admin";
const char* www_password = "http_auth";

void setup()
{
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  Serial.print("\nStarting HTTPBasicAuth_SAMD_M0 on SAMD Zero with"); Serial.println(SHIELD_TYPE);
  Serial.println(ETHERNET_WEBSERVER_VERSION);

  ET_LOGWARN(F("SPI pinout:"));
  ET_LOGWARN1(F("MOSI:"), MOSI);
  ET_LOGWARN1(F("MISO:"), MISO);
  ET_LOGWARN1(F("SCK:"),  SCK);
  ET_LOGWARN1(F("CS:"),   USE_THIS_SS_PIN);
  ET_LOGWARN(F("========================="));

  Ethernet.init (USE_THIS_SS_PIN);
    
  // Use DHCP
  Ethernet.begin(mac);
  // Use Static IP
  //Ethernet.begin(mac, ip); 
  // Use static IP and DNS
  //Ethernet.begin(mac, ip, myDns);

  Serial.print(F("Connected! IP address: "));
  Serial.println(Ethernet.localIP());

  server.on(F("/"), []()
  {
    if (!server.authenticate(www_username, www_password))
    {
      return server.requestAuthentication();
    }
    server.send(200, F("text/plain"), F("Login OK"));
  });

  server.begin();

  Serial.print(F("Open http://")); Serial.print(Ethernet.localIP());
  Serial.println(F("/ in your browser to see it working"));
  Serial.print(F("Using username : ")); Serial.print(www_username);
  Serial.print(F(" and password : "));  Serial.println(www_password);
}

void loop()
{
  server.handleClient();
}
```

I'm OK here. Be sure to select `SAMD => Arduino Zero (Native USB Port)`

```
Starting HTTPBasicAuth_SAMD_M0 on SAMD Zero withW5x00 using Ethernet Library with Large Buffer
EthernetWebServer v2.2.3
[EWS] SPI pinout:
[EWS] MOSI: 23
[EWS] MISO: 22
[EWS] SCK: 24
[EWS] CS: 10
[EWS] =========================
Connected! IP address: 192.168.2.226
Open http://192.168.2.226/ in your browser to see it working
Using username : admin and password : http_auth
```

proasnet, 2022-09-27:
hi @khoih-prog,
this example working.
Thank you.
Where was a problem?

khoih-prog, 2022-09-27:
Missing

```
// Default to use W5100. Must change to false for W5500, W5100S, for faster SPI clock
// Must use true for SAMD21, such as Zero, SAMD_FEATHER_M0_EXPRESS, etc.
#define USE_W5100                           true
```

That's why the library tried to use the more aggressive SPI timings for newer boards, and SAMD Zero can't take it.

proasnet, 2022-09-27:
Dear @khoih-prog , if this working fine, I trying find example with some form, for example as in old library, that I used in my project and I am reworking all for your library - setting ethernet parameters by web, setting other program values stored in flash.

After login to server, user or admin can display web form with traditional network settings, that is saved to eeprom in the mega case, with SAMD M0 to flash.
Can you recomed me some example with similar form, please? In your examples I can not find some POST forms.
Thanks.

khoih-prog, 2022-09-27:
HI @proasnet 

You can check the POST-related examples in this library, such as 

1. [**PostServer**](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/PostServer)
2. [**SimplePost**](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/HTTPClient/SimplePost)

If you need help, I suggest that you do some little Google-research, or post the issue on `Arduino Forum`, to ask for help from many experts there. I'm sorry I wish I have time helping and dealing with personal project.

Good Luck,