khoih-prog/EthernetWebServer #66: RP2040 (IP Unset) with localIP

KBLiveSolutions, 2022-09-22:
Hi

I am running the UdpSendReceive with a RP2040 (Earle Philhower port) and a W5500 Ethernet Shield 
This is what I get when running the code

`
Start UDPSendReceive on RASPBERRY_PI_PICO with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer v2.2.3
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 17
[ETG] W5100 init, using W5100Class::ss_pin =  17 , whereas new ss_pin =  10 , SS_PIN_DEFAULT = 17
[ETG] Chip is W5500
[ETG] W5100::init: W5500, SSIZE = 8192
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
Using mac index = 17
Connected! IP address: (IP unset)

Starting connection to server...
Listening on port 1883`

Why do I have this **(IP unset)** for the localIP ?
The Ethernet shield seems to be correctly connected.

khoih-prog, 2022-09-22:
Sorry I have no issue here.

```
Start UDPSendReceive on RASPBERRY_PI_PICO with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer_SSL v1.9.2
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 17
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
Using mac index = 9
Connected! IP address: 192.168.2.108
```

Check your router / wiring, especially Ethernet cable 

This happens if I **remove the Ethernet cable**, then reset

```
Start UDPSendReceive on RASPBERRY_PI_PICO with W5x00 using Ethernet_Generic Library on SPI0/SPI
EthernetWebServer_SSL v1.9.2
[EWS] =========== USE_ETHERNET_GENERIC ===========
[EWS] Default SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
[EWS] RPIPICO setCsPin: 17
[EWS] =========================
[EWS] Currently Used SPI pinout:
[EWS] MOSI: 19
[EWS] MISO: 16
[EWS] SCK: 18
[EWS] SS: 17
[EWS] =========================
Using mac index = 18
Connected! IP address: (IP unset)

Starting connection to server...
Listening on port 1883
```

closed 2022-09-22 by khoih-prog

KBLiveSolutions, 2022-09-22:
Ok, I'll check the cables, thanks